openhwfoundation/cva6 #2383: [BUG] <title> Errors including undefined reference to `__atomic_compare_exchange_4' when running "make images" command

walrider3, 2024-07-23:
### Is there an existing CVA6 bug for this?

- [X] I have searched the existing bug issues

### Bug Description

I'm trying to run the "make images" so I can make a bootable SD card. However, it is not working since I get the following error:/mnt/c/Users/PMLS/Downloads/cva6_sdk_master/buildroot/output/host/bin/riscv64-buildroot-linux-uclibc-nm: libc/stdlib/__cxa_finalize.os: file format not recognized
  AR cr lib/libc.a
  STRIP -x -R .note -R .comment lib/libc.a
  CC libpthread/nptl/pthread_atfork.oS
  AR cr lib/libpthread_nonshared.a
  STRIP -x -R .note -R .comment lib/libpthread_nonshared.a
  LD libuClibc-1.0.49.so
/mnt/c/Users/PMLS/Downloads/cva6_sdk_master/buildroot/output/host/lib/gcc/riscv64-buildroot-linux-uclibc/13.3.0/../../../../riscv64-buildroot-linux-uclibc/bin/ld: libc/libc_so.a(err.os): in function `.L0 ':
err.c:(.text+0xa0): undefined reference to `__atomic_compare_exchange_4'
/mnt/c/Users/PMLS/Downloads/cva6_sdk_master/buildroot/output/host/lib/gcc/riscv64-buildroot-linux-uclibc/13.3.0/../../../../riscv64-buildroot-linux-uclibc/bin/ld: libc/libc_so.a(err.os): in function `.L6':
err.c:(.text+0x138): undefined reference to `__atomic_exchange_4'
/mnt/c/Users/PMLS/Downloads/cva6_sdk_master/buildroot/output/host/lib/gcc/riscv64-buildroot-linux-uclibc/13.3.0/../../../../riscv64-buildroot-linux-uclibc/bin/ld: libc/libc_so.a(fork.oS): in function `.L23':
fork.c:(.text+0x130): undefined reference to `__atomic_fetch_add_4'
/mnt/c/Users/PMLS/Downloads/cva6_sdk_master/buildroot/output/host/lib/gcc/riscv64-buildroot-linux-uclibc/13.3.0/../../../../riscv64-buildroot-linux-uclibc/bin/ld: libc/libc_so.a(register-atfork.oS): in function `.L0 ':
register-atfork.c:(.text+0x4c): undefined reference to `__atomic_compare_exchange_8'
/mnt/c/Users/PMLS/Downloads/cva6_sdk_master/buildroot/output/host/lib/gcc/riscv64-buildroot-linux-uclibc/13.3.0/../../../../riscv64-buildroot-linux-uclibc/bin/ld: libc/libc_so.a(sem_timedwait.oS): in function `.L5':
sem_timedwait.c:(.text+0xd8): undefined reference to `__atomic_fetch_add_8'
collect2: error: ld returned 1 exit status
make[2]: *** [libc/Makefile.in:77: lib/libc.so] Error 1
make[2]: Leaving directory '/mnt/c/Users/PMLS/Downloads/cva6_sdk_master/buildroot/output/build/uclibc-1.0.49'
make[1]: *** [package/pkg-generic.mk:289: /mnt/c/Users/PMLS/Downloads/cva6_sdk_master/buildroot/output/build/uclibc-1.0.49/.stamp_built] Error 2
make[1]: Leaving directory '/mnt/c/Users/PMLS/Downloads/cva6_sdk_master/buildroot'
make: *** [Makefile:77: /mnt/c/Users/PMLS/Downloads/cva6_sdk_master/buildroot/output/host/bin/riscv64-buildroot-linux-gnu-gcc] Error 2

I don't understand what to do. I traced the files mentioned in the errors and tried to find out what was wrong but to now avail.
I would greatly appreciate any help.

walrider3, 2024-07-23:
Ummm, any help???

Moschn, 2024-07-23:
The issue is not even 6 hours old and you are already (aggressively) asking for help. Most of us here providing help do so voluntarily. You are not entitled to a quick response or any response at all. 

Nevertheless, let us get back to your issue. What is this `make images` command and why are you using it? Is this related to cva6-sdk? If yes, you should close this issue and repost it in the correct repository.

walrider3, 2024-07-23:
> The issue is not even 6 hours old and you are already (aggressively) asking for help. Most of us here providing help do so voluntarily. You are not entitled to a quick response or any response at [...]

Sorry, I didn't mean it in a rude way, I've been working on this for past week and was just very sick of this. But I did not mean at all to be rude or demanding rather it was something I wrote on a whim.
About the issue, the "make images command" is part of the of readme instruction on the ariane-sdk repository. It is basically to create the fw_payload.bin file that includes (opensbi +uboot), and also makes the linux image to flash onto the sd card. However, running the command gives the above errors. I've tried searching quite a lot for a solution or related problem at least but to no avail. In hindsight, I guess I should have opened the issue on the ariane-sdk repo but I figured who has run CVA6 must have also run the "make images" since we can't boot Linux on CVA6 without that.

JeanRochCoulon, 2024-07-23:
Please add a Github issue in ariane-sdk if you need. I close this issue.

closed 2024-07-23 by JeanRochCoulon