openhwfoundation/cva6 #2392: [BUG] wrong pmp entries length

AyoubJalali, 2024-07-24:
### Is there an existing CVA6 bug for this?

- [X] I have searched the existing bug issues

### Bug Description

this is a RTL Bug, I notice that in [CSR_REGFILE](https://github.com/openhwgroup/cva6/blob/master/core/csr_regfile.sv), it been declare a table of [pmpcfg_t ](https://github.com/openhwgroup/cva6/blob/846e1a1269cf1e8ca3573de06b299a9e2e5b7b5d/core/csr_regfile.sv#L166)(representing the number of pmp entries supported), also in [pmpaddr](https://github.com/openhwgroup/cva6/blob/846e1a1269cf1e8ca3573de06b299a9e2e5b7b5d/core/csr_regfile.sv#L168), and it's been declare like that [[CVA6Cfg.]](riscv::pmpcfg_t [CVA6Cfg.NrPMPEntries:0] pmpcfg_o) this create 9 pmp entries and not 8 (there's a -1 missing ) also in line 168 in the same file

Moschn, 2024-07-25:
Good catch! I am preparing a PR to fix this.

closed 2024-07-25 by JeanRochCoulon

AngelaGonzalezMarino, 2024-11-08:
Now this fails if NrPMPEntries=0 as in FPGA configuration
I will fix it in the Altera support PR, making it 0 in that case and not -1 as it is now

PR #2394 (2024-07-25): Fix off by one error in PMP length
Fixes #2392