openhwfoundation/cva6 #2400: [BUG] NAPOT MODE isn't supported

AyoubJalali, 2024-07-25:
### Is there an existing CVA6 bug for this?

- [X] I have searched the existing bug issues

### Bug Description

Hello, according to the [cv32a65x](https://docs.openhwgroup.org/projects/cva6-user-manual/04_cv32a65x/riscv/priv.html#_machine_status_mstatus_and_mstatush_registers), in the pmp entries the NA4 & NAPOT modes isn't supported, but in the RTL is only excluding NA4 from the add_mode logic in [csr_regfile.sv](https://github.com/openhwgroup/cva6/blob/fd489a16fbfbb791d598c4a61afa765267ab12ff/core/csr_regfile.sv#L2690)

is it a bug in the RTL or the Specification ?

AyoubJalali, 2024-08-29:
I think this is a RTL bug

zchamski, 2024-09-23:
Results of a careful review of the spec and RTL with @ASintzoff:
- the spec of CV32A65X **does not specify** the actual WARL behavior upon writing an invalid (illegal/reserved) multi-bit value for RWX and A fields; this is required given the definition of WARL at https://cva6.readthedocs.io/en/latest/04_cv32a65x/riscv/priv.html#_write_any_values_reads_legal_values_warl;
- the RTL implements 'no change on reserved' policy for the aggregate RWX field;
- the RTL implements 'no change on illegal' policy for A fields but **does** allow `A == NAPOT` which conflicts with the CV32A65X spec  supporting only `A == OFF` and `A == TOR`.

The RTL needs to be fixed for CV32A65X but without affecting other configurations.

commit 2155d0e9 (2025-01-07): Fix #2665 #2400 #2657 (#2685)

* Fix #2400: define new PmpNapotEn parameter to disable NAPOT

* Fix #2665 by removing NA4 related RTL lines

* Fix Spyglass

* Fix gate count
closed 2025-01-07 by ASintzoff