openhwfoundation/cva6 #2413: Porting CVA6 to VCU118: SPI Signal Issues

mrbilandi, 2024-07-29:
I am going to port CVA6 to the VCU118 board. I have synthesized the FPGA code for the VCU118 and used the PMOD micro SD card adapter from Digilent. However, I observed that Linux gets stuck at a certain point. After troubleshooting with an oscilloscope, I noticed that the clock signal of the SPI interface only goes up to 1.2V, whereas it should reach 3.3V, which is the operating voltage of the SD card.
[image]

After checking the schematic, I saw that Xilinx used a level translator to convert the FPGA voltage bank from 1.2V to 3.3V and vice versa. The reference voltage of this translator was connected to 1.2V, but the voltage of the FPGA's bank that this translator was connected to was 1.8V. I thought this mismatch might be the cause, so I changed the bank voltage to 1.2V using a software called "VCU118 SCUI" provided by Xilinx. However, the clock signal remained the same.

After researching this issue, I discovered that several people have reported the same problem with the PMOD port on the VCU118 when used for the SD card. The issue is that this translator is open-drain and requires pull-up resistors at its output. The adapter we bought has these pull-up resistors, but their value is too high (20 kOhms), causing the clock signal to rise very slowly. Therefore, we need to replace these resistors with lower-value ones. There is a recommendation online to use resistors around 470 ohms.

So overall, I think I should do two things: firstly, decrease the frequency of the SPI interface to see if the rise time of the clock signal improves. secondly, if the first step doesn't work, replace these pull-up resistors with lower-value ones.

My question is: How can I decrease the frequency of the SPI?
I have increased the "Frequency Ratio" in the xlnx_axi_quad_spi module, but I'm not sure if I also need to adjust the frequency in the boot-up firmware.

Thanks

mrbilandi, 2024-07-29:
Hi [dassheladiya](https://github.com/dassheladiya),

I actually took a different approach. I built the FPGA code of CVA6 for the Genesys2 board and then modified it in Vivado (GUI) for the VCU118 instead of modifying the makefile and TCL scripts. I hope I can do that later. For now, there are two USB-UART ports. Are you sure you selected the correct one in the terminal?

I used these pins in the .xdc file which is connected to port2 of USB-UART:

set_property -dict {PACKAGE_PIN AW25 IOSTANDARD LVCMOS18} [get_ports rx];
set_property -dict {PACKAGE_PIN BB21 IOSTANDARD LVCMOS18} [get_ports tx];

Now I can see that the boot ROM works but gets stuck when it tries to get data from the SD card, as you can see below.

Hello World!
init SPI
status: 0x0000000000000025
status: 0x0000000000000025
SPI initialized!
initializing SD... 
could not initialize sd... exiting

mrbilandi, 2024-07-30:
In the bootloader, there are some "nop" instructions after SPI transmit. If we increase the frequency ratio, we need to increase these delays implemented using "nop" instructions. After increasing these delays, the bootloader can communicate with the SD card and read from it, but it is too slow!

mrbilandi, 2024-08-01:
I added some printf statements in U-Boot to see what is going on, and it turns out that when it wants to read data with a length of more than 256 bytes, the next byte is lost and consequently the CRC fails. I changed the maximum block size (MMC_MAX_BLOCK_LEN) to 256. I saw that the retrieved data is correct, but the calculated CRC fetched from MMC is not the same (mmc_spi_readdata: data CRC error, expected 5384 got 988f).

I got stuck at this step and would really appreciate it if anyone can help.

mrbilandi, 2024-08-02:
Okay, I got it. Since I decreased the SPI frequency, it is necessary to add a delay in the SPI driver for U-Boot to be able to read data properly from the SD card, just like I did for the boot ROM. After adding this delay, I now have Linux booting up successfully without changing the pull-up resistors. However, it's slow; it takes around 5 minutes to boot Linux!

dassheladiya, 2024-08-02:
Hi @mrbilandi ,
Congratulations on successfully running Linux. I have selected the correct UART port, but my screen remains blank. Is your work open source? If so, could you kindly share the changes in a patch file, if possible?

closed 2024-09-08 by github-actions[bot]