openhwfoundation/cva6 #2416: Define Memory Access Type

AEzzejjari, 2024-07-29:
### Is there an existing CVA6 bug for this?

- [X] I have searched the existing bug issues

### Bug Description

The type of memory access is information we need to check other CVA6 components. 
I'd like to know if there's a way, from the current AXI interface, to provide information on whether we're fetching or not.

JeanRochCoulon, 2024-07-30:
If I understand, the read and write information is present, but the fetch is not present on AXI.

On another hand, what is the status of OBI ?

cfuguet, 2024-07-30:
Hello, I'm not sure to understand the question. Was the original question @AEzzejjari if there is a way to see if a given AXI request is an instruction cache miss ?

Regarding OBI, I need to make a meeting with Akiho to see how is she progressing. I will let you know as soon as I have some news.

JeanRochCoulon, 2024-07-31:
To me, @AEzzejjari is looking for an information from AXI bus to know whether an access is a fetch or not. Can you confirm @AEzzejjari ?

cfuguet, 2024-07-31:
Ok, if that is the question, then yes, there is a specific AXI ID for Icache miss transactions.

```systemverilog
core/cache_subsystem/cva6_hpdcache_subsystem.sv:148:localparam int ICACHE_RDTXID = 1 << (CVA6Cfg.MEM_TID_WIDTH - 1);
```

AEzzejjari, 2024-10-01:
But we cannot use this information in the testbench.

JeanRochCoulon, 2025-01-02:
@AEzzejjari We are cleaning the "Github Issues". I assume this issue is closed. May I ask you to close it if it is the case (with a comment please) ?