openhwfoundation/cva6 #2423: Simplest way to run a U-mode binary with virtual memory

tunefish777, 2024-07-30:
Hi,
I have recently returned to working with the CVA6 core and quite a lot has changed since I last tinkered around with it.
Previously running simple U-mode binaries was very trivial with Verilator and the RISC-V PK, but the latter is now deprecated, everything points towards OpenSBI and the current version of the README only explains baremetal applications and regression/smoke-tests.
So my question is pretty simple: What is the best path to achieve what was previously possible again?
Booting a Linux image is not an option but I need some OS that sets up basic virtual memory and starts a U-mode application

Cheers,
Moritz

JeanRochCoulon, 2024-07-31:
Riscv-pk is no more supported by cva6.

Moschn, 2024-07-31:
Currently, there is no supported way to do what you want or what was once possible. As far as I remember, maintaining riscv-pk and the many problems that users had to get it running led to it being no longer officially supported by CVA6. 

I believe there should not be any technical limitation that renders it impossible to run binaries using riscv-pk on CVA6, but you will have to get your hands dirty. In case you get it to work again, we might be interested to look into officially supporting it again.

closed 2024-09-05 by github-actions[bot]