openhwfoundation/cva6 #2425: cva6 Synthesis

Syedateeb24, 2024-07-31:
### Is there an existing CVA6 bug for this?

- [X] I have searched the existing bug issues

### Bug Description

I am trying to perform synthesis of cva6 on cadence genus but i am unable to perform a successful synthesis. There are many design files in the repository which is posing a problem for me to choose the required/necessary files for synthesis. I request you to help me know the right way to approach the synthesis & help me in knowing the necessary files to perform a successful synthesis of cva6.

Syedateeb24, 2024-08-01:
Any inputs on synthesis will be highly appreciated.
Thankyou

Moschn, 2024-08-02:
Take a look at [core/Flist.cva6](https://github.com/openhwgroup/cva6/blob/0c60bc6e3dccd61753aeb2176fc1d586b283578b/core/Flist.cva6). It should contain most files required for synthesis. 

You can also check the Makefile for more details.

hhhsiang, 2024-10-10:
hi @Syedateeb24 , can you synthesize the cva6 core successfully?