openhwfoundation/cva6 #2426: How to bring printf support in simulation environment ?

isaar-ahmad, 2024-07-31:
### Is there an existing CVA6 bug for this?

- [X] I have searched the existing bug issues

### Bug Description

Hi,
I found that there has been recent discussion (May 2024 - July 2024) regarding printf statement support in CVA6 simulations (see #2103 #2184 #2200 #2220 ). It seems that CVA6 simulation environment, in its current form (e.g. with Verilator), doesn't support this. 

For example, Dhrystone C files [1] have debug_print statements in them. To a newcomer, this would give an initial impression that these prints should be visible somewhere in the logfiles or console output. However, a closer inspection reveals that the implementation of debug_print() function is empty. In development stage, it is common in bare-metal environments to disable print statements, so that other parts of the test can pass through.

That being said, printf statements are useful in debugging, especially when the target under debug is a CPU itself. (I only speak for myself, based on my experience with RTL verification of ARM CPUs, and some RISC-V CPUs as well).

Please advise on this : 
1. Is there any ongoing (or planned) development for C print statements ? If yes, where can we find details regarding this ? 
3. What parts of the testbench and RTL files will need major changes, if such an update is planned?

Thanks,
Isaar 

**Code reference :** 
[1] Dhrystone source files (dhrystone_main.c and dhrystone.c) : Fetched by [install-riscv-tests.sh](https://github.com/openhwgroup/cva6/blob/master/verif/regress/install-riscv-tests.sh)

JeanRochCoulon, 2024-07-31:
Indeed this feature is highly requested. But as you know it is not supported. The solution would be to support pk which provides the print feature. If someone is ready to work on it, we can provide recommendations.

isaar-ahmad, 2024-08-07:
Yes, please share a desirable plan (or an outline) for the required functionality. That would be helpful in assessing whether someone (e.g. myself) can take up the task.

JeanRochCoulon, 2024-08-07:
@valentinThomazic the floor is yours to explain how to proceed ;-)

valentinThomazic, 2024-08-29:
Hey sorry for the delay, the riscv proxy kernel used to be supported in this repo (I can give some directions if needed).
For it to be supported again, we would need it to be cleanly integrated to the flow. 
It should be possible to activate it with an env variable or a parameter for instance. 
The difficulty would come from the complexity of the `cva6.py `script, the `verif/sim/Makefile` and `./Makefile`, we really don't want to add more.

jason23g, 2024-11-12:
Hello @valentinThomazic ,

Could you please give some directions to be able to support riscv proxy kernel in the current environment of cva6. I tried to install riscv pk from the corresponding repository of riscv but when i run the executable of Variane_testharness along with the proxy kernel and the desirable executable ( in my case an elf file of hello world in order to see the print happening in my terminal) all i get is an Assertion Failed  "ariane_testharness.sv:671: Assertion failed in TOP.ariane_testharness.p_assert: B Response Errored".  In addition i would like to ask if the riscv proxy kernel after it had been built will it  support to run in spike tandem mode ?

Thanks in advance for your time

valentinThomazic, 2024-11-12:
Hey @jason23g,

You may have a compilation options issue. For instance if you are trying to run the pk on one of the 32 bits configuration of the CVA6, were the pk and the hello_world compiled for 32 bits ? 

This may help you : https://github.com/openhwgroup/cva6/issues/1358#issuecomment-1808134071 as it will work on the Makefile default's configuration. Please note that we will not provide support for using `make verilate` command flow. This is given only for testing and potential integration into the current flow.

jason23g, 2024-11-12:
Hello @valentinThomazic ,

The pk and the hello world were compiled for 64 bits with the flow of the comment you gave in [#1358](https://github.com/openhwgroup/cva6/issues/1358#issuecomment-1808134071) but still doesn't work. Could you please give some directions to be able to support riscv proxy kernel in the current environment of cva6.

Thanks in advance for your time

valentinThomazic, 2024-11-12:
Hey @isaar-ahmad and @jason23g 

1&4 => If you take a look at `cva6.py` you can see that there are some flaws. For instance, there are 3 functions to run directed tests depending on the extension of the program to run which are basically copy pasted from each others. One of them is never used (the elf one) and I am not sure that it currently works. Because of that, redundant work is sometime needed and it is the cause of issues and errors. IIRC some functionnalities are broken like running several programs from a directory.
Also the simulation flow is quite complex : simulations are ran through `cva6.py` which uses `verif/sim/Makefile` which uses `./Makefile`.

I am not asking you to clear and refactor `cva6.py` or the simulation flow but it is just to explain that when I was talking about integrating the functionality cleanly I was asking you to be precautious with the added complexity.

2 => You will probably need to change `cva6.py`, `verif/sim/Makefile` and `./Makefile`. 

3 => This is how it would have been done previously : https://github.com/openhwgroup/cva6/issues/1358#issuecomment-1808134071, pk was dropped with : https://github.com/openhwgroup/cva6/pull/1663 but the instructions were already broken so you should take a look at the issue comment instead. Basically the compilation and the pk and the program were done manually, it would need to be automated:

1. The user enable an env variable (a `cva6.py` arg could also be used)
2. A simulation is run by the user (with `cva6.py`)
3. The pk and the program are compiled for the configuration on which the simulation is run
4. Both the pk executable and the program are passed to the simulator

This should work for all configurations and simulators combination and if possible with Spike Tandem.

Hope this helps.

isaar-ahmad, 2025-01-06:
@valentinThomazic @JeanRochCoulon I have started working on this. Please bear with me, as the response is a bit lengthy. 

I took the approach suggested in https://github.com/openhwgroup/cva6/issues/1358#issuecomment-1808134071 and still faced issues.

Summary of current status :  
I made a minimal set of changes to current makefiles and cva6.py . After applying necessary changes, the simulation exited with an 'Illegal instruction was executed' exception. I found that this was printed from riscv-pk.

Details : 

**Step 1 : CVA6 and RISCV-PK setup for cv64_imafdc_sv39 target with arch `rv64gc_zba_zbb_zbs_zbc_zicsr_zifencei` and abi `lp64d`**

1. Downloaded the latest clone of CVA6 repo (commit 86a80f0e) .
2. For RISCV toolchain, I used a previous installation (downloaded in July 2024). 
3. Cloned the latest version of riscv-pk. Configured it to associate with the RISCV toolchain in (2).
`./configure` --prefix=$RISCV --host=riscv-none-elf --with-arch=rv64gc_zba_zbb_zbs_zbc_zicsr_zifencei --with-abi=lp64d`

The choice of `with-arch` was dictated by what is currently supported in verif/sim/cva6.py .My idea was to ensure consistency between riscv-pk and the supported CVA6 64-bit implementation.

4. Edited cva6.py to ensure that we can compile with zifencei and zicsr .

**Step 2: Prepared a hello_world test script with `isa_extension=zicsr,zifencei`.** 

The script was placed in verif/sim/ folder. It looks similar to verif/regress/smoke-tests-cv64a6_imafdc_sv39.sh , with some adaptation to the `python cva6.py ... ` part.
 
`python3 cva6.py --target cv64a6_imafdc_sv39 --iss=$DV_SIMULATORS --iss_yaml=cva6.yaml **--iss_timeout=2000** \
--c_tests ../tests/custom/hello_world/hello_world.c **--isa_extension="zicsr,zifencei"** --linker=../../config/gen_from_riscv_config/linker/link.ld \
--gcc_opts="-static -mcmodel=medany -fvisibility=hidden -nostdlib \
-nostartfiles ../tests/custom/common/syscalls.c \
../tests/custom/common/crt.S -lgcc \
-I../tests/custom/env -I../tests/custom/common"`

**Step 3 : Simplifying veri-testharness section in verif/sim/Makefile (without invoking RISCV-PK) :** 

I commented the following section of veri-testharness target in verif/sim/Makefile : 

`make -C $(path_var) verilate verilator="verilator --no-timing" target=$(target) defines=$(subst +define+,,$(isscomp_opts))`
` $(path_var)/work-ver/Variane_testharness $(if $(TRACE_COMPACT), -f verilator.fst) $(if $(TRACE_FAST), -v verilator.vcd) $(elf) $(issrun_opts) $(COMMON_PLUS_`ARGS)`

I simplified it to : 

`make -C $(path_var) verilate`
`$(path_var)/work-ver/Variane_testharness $(if $(TRACE_COMPACT), -f verilator.fst) $(if $(TRACE_FAST), -v verilator.vcd) $(elf)`

This version of `veri-testharness` passed with hello world script (see Step 3). So, I went to next step.

**Step 4 : Adding RISCV-PK to veri-testharness section :** 

`make -C $(path_var) verilate`
`$(path_var)/work-ver/Variane_testharness $(if $(TRACE_COMPACT), -f verilator.fst) $(if $(TRACE_FAST), -v verilator.vcd) $(RISCV)/riscv-none-elf/bin/pk $(elf) \
                +core_name=$(target)`
 
This failed at the simulation stage. The logfile error message looks as follows : 
`
<cva6_repo_dir>/cva6_jan0325//work-ver/Variane_testharness  -f verilator.fst  <riscv_home_dir>/riscv/riscv-none-elf/bin/pk 
<cva6_repo_dir>/cva6_jan0325/verif/sim/out_2025-01-06/directed_tests/hello_world.o \
	+core_name=cv64a6_imafdc_sv39
This emulator compiled with JTAG Remote Bitbang client. To enable, use +jtag_rbb_enable=1.
Listening on port 32865
Starting FST waveform dump into file 'verilator.fst'...
z  0000000000000000 ra 0000000000000000 sp 0000000000000000 gp 0000000000000000
tp 0000000000000000 t0 000000000001e000 t1 0000000000000000 t2 0000000000000000
s0 0000000000000000 s1 0000000000000000 a0 0000000000000000 a1 0000000000000000
a2 0000000000000000 a3 0000000000000000 a4 0000000000000000 a5 0000000000000000
a6 0000000000000000 a7 0000000000000000 s2 0000000000000000 s3 0000000000000000
s4 0000000000000000 s5 0000000000000000 s6 0000000000000000 s7 0000000000000000
s8 0000000000000000 s9 0000000000000000 sA 0000000000000000 sB 0000000000000000
t3 0000000000000000 t4 0000000000000000 t5 0000000000000000 t6 0000000000000000
pc 0000000080000040 va/inst 000000003002a073 sr 8000000200006020
An illegal instruction was executed!
<riscv_home_dir>/riscv/riscv-none-elf/bin/pk *** FAILED *** (tohost = -1) after 2296061 cycles
*** [rvf_tracer] INFO: Loading binary : 
*** [rvf_tracer] INFO: tohost_addr: 0000000000000000
*** [rvf_tracer] WARNING: No valid address of 'tohost' (tohost == 0x0000000000000000), termination possible only by timeout or Ctrl-C!

CPU time used: 819232.98 ms
Wall clock time passed: 820683.46 ms
make: *** [Makefile:422: veri-testharness] Error 255`

**Step 5: Debugging**

The failure message came from riscv-pk. The corresponding instruction (address 0x80000040) was `csrs mstatus, t0` (see disassembly of hello_wolrd.o below). I found that this error maybe due to incorrect privilege mode  : 
[image]

@valentinThomazic @JeanRochCoulon This is where we are. Where do we go from here?

valentinThomazic, 2025-01-11:
Hello @isaar-ahmad,

I have a few questions :
- Did you manage to run the hello world test with the pk successfully at some point? 
- Also, did you try to run your proxy kernel without any elf, to see if it works with the CVA6?
- Which compilations options did you use to compile the hello world tests ? If you used a smoke tests script as an example, some options should be changed. For instance, we do not want the -nostdlib flag in this case. IIRC I had success with basically no compilations options except for the -march and -mabi. Please take a look at the previously mentionned comment with instructions.

Also, I think it may be better to create a new target for the pk instead of using veri-testharness because by doing so, all simulations with this target would run with pk, which we do not want. It could be something like veri-testharness-pk.
Regarding the simplified version of the target, I am not sure we actually want to simplify it. For instance, user are supposed to be able to pass options to verilator by calling cva.py with the issrun_opts parameter.

gcchri, 2025-01-17:
Hello @valentinThomazic and @isaar-ahmad. Thank you for your efforts in this. I managed to print (from both supervisor and user mode) using the flow from https://github.com/openhwgroup/cva6/issues/1358#issuecomment-1808134071. However, it does not work with the latest pk version, rather with [this](https://github.com/riscv-software-src/riscv-pk/tree/710c23a5bbeecf171ac86d6e39d275af8f176354) version. Additionally, I added `+tohost_addr=8000c000` in the `Variane_testharness` command. The simulation fails at the end, but at least there is print functionality. I tried the steps in https://github.com/openhwgroup/cva6/issues/2426#issuecomment-2573912073 on my setup but could not get it to either, it seems that some gcc flags are used automatically. I will post updates if I have any success.

isaar-ahmad, 2025-01-17:
Hi @valentinThomazic ,

I got the simplest example working after factoring your recent inputs.  I used a recent version of RISC-V pk (available around January 3 2024) and compiled it with the steps in https://github.com/openhwgroup/cva6/issues/1358#issuecomment-1808134071

I was able to run hello.c example described in https://github.com/openhwgroup/cva6/issues/1358#issuecomment-1808134071

So, basically the following worked : 
```
cd $REPO_HOME/verif/sim

$RISCV/bin/riscv-none-elf-gcc -march=rv64imafdc_zicsr_zifencei -mabi=lp64d hello.c -o hello.elf

../../work-ver/Variane_testharness <absolute_path_to_riscv_build>/riscv/riscv-none-elf/bin/pk hello.elf
```
[image]

I

@gcchri , please see if you are able to reproduce this as well.

**Next steps ** 
Based on @valentinThomazic 's suggestion, I am now working on adding a separate veri-testharness-pk target. 

For this, I have : 
(1) added a veri-testharness-pk target to verif/sim/Makefile and verif/sim/cva6.yaml 
(2) few lines to verif/sim/cva6.py (to ensure that default GCC compilation options are not applied to veri-testharness-pk)

If you would like to see those changes and provide suggestions, let me know how to share those? Shall I open a pull-request, or share with you privately?

gcchri, 2025-01-17:
I manage to print with the latest pk but it still fails. I do not get the "No explicit FST file name supplied, using RTL defaults." printout, if that makes any difference.

isaar-ahmad, 2025-01-17:
@gcchri could you share a snippet from your logfiles ? I also had failures, and figured it was due to simulation timeout. I guess it may apply to your case as well (?)

There are two timeout parameters : (1) for wall simulation time (default 600 seconds), (2) max clock cycles (default value 2000000) 

For (1), use the --iss_timeout parameter (passed to cva6.py).

For (2), apply the following patch to increase the max simulated clock cycles : 

diff --git a/corev_apu/tb/rvfi_tracer.sv b/corev_apu/tb/rvfi_tracer.sv
index b48a4836..141e16c0 100644
--- a/corev_apu/tb/rvfi_tracer.sv
+++ b/corev_apu/tb/rvfi_tracer.sv
@@ -43,7 +43,7 @@ module rvfi_tracer #(
   initial begin
     TOHOST_ADDR = '0;
     f = $fopen($sformatf("trace_rvfi_hart_%h.dasm", HART_ID), "w");
-    if (!$value$plusargs("time_out=%d", SIM_FINISH)) SIM_FINISH = 2000000;
+    if (!$value$plusargs("time_out=%d", SIM_FINISH)) SIM_FINISH = 20000000;
     if (!$value$plusargs("tohost_addr=%h", TOHOST_ADDR)) TOHOST_ADDR = '0;
     if (TOHOST_ADDR == '0) begin
         if (!$value$plusargs("elf_file=%s", binary)) binary = "";

I increased both values to rule out this issue.

Thanks,
Isaar

gcchri, 2025-01-17:
Thank you. Worked like a charm. I just changed the max clock cycle count, since it takes less that 10 minutes to run a simple hello world. It was so close to the default 200k cycles though (~205k).

isaar-ahmad, 2025-01-24:
Update : 

I am able to run complex programs with printf enabled.
I will be opening a pull request within this week (unless someone else beats me to it).

Here is what I am planning to do. @JeanRochCoulon and @valentinThomazic , please provide your feedback on this. If anything is missing here, let me know : 
1. Installation script for installing pk : pk may need to be reconfigured every time a user simulates a different target (e.g. switching from 32-bit variant to 64-bit variant).
2. verif/sim/Makefile : Add 'veri-testharness-pk' target. This will directly 
3. verif/sim/cva6.yaml : Define 'veri-testharness-pk' target in a similar manner to 'veri-testharness'.
 
4. cva6.py : Add an if-else block inside run_test() function, so that certain .sv files are skipped for 'veri-testharness-pk'. It will look somewhat like below : 
```
diff --git a/verif/sim/cva6.py b/verif/sim/cva6.py
index a2a37654..4c211278 100644
--- a/verif/sim/cva6.py
+++ b/verif/sim/cva6.py
@@ -501,13 +501,18 @@ def run_test(test, iss_yaml, isa, target, mabi, gcc_opts, iss_opts, output_dir,
   if test_type != "o":
     # gcc compilation
     logging.info("Compiling test: %s" % test_path)
-    cmd = ("%s %s \
-          -I%s/dv/user_extension \
-            -T%s %s -o %s " % \
-          (get_env_var("RISCV_CC", debug_cmd = debug_cmd), test_path, cwd,
-              linker, gcc_opts, elf))
+    if "veri-testharness-pk" not in iss_list:
+        cmd = ("%s %s \
+              -I%s/dv/user_extension \
+                -T%s %s -o %s " % \
+              (get_env_var("RISCV_CC", debug_cmd = debug_cmd), test_path, cwd,
+                  linker, gcc_opts, elf))
+    else:
+        #cmd= ("%s %s -o %s " % (get_env_var("RISCV_CC", debug_cmd = debug_cmd), test_path, elf))
+        cmd= ("%s %s %s -o %s " % (get_env_var("RISCV_CC", debug_cmd = debug_cmd), test_path, gcc_opts, elf))
     cmd += (" -march=%s" % isa)
     cmd += (" -mabi=%s" % mabi)
+    logging.info("Compilation command : %s" % cmd)
     run_cmd(cmd, debug_cmd = debug_cmd)
```
Thanks,
Isaar

JeanRochCoulon, 2025-01-24:
Hello @isaar-ahmad 
This is a great contribution.
I am in line with your plan.
@valentinThomazic turns back mid-feb, and will be ready to review your PR.

valentinThomazic, 2025-02-17:
Hello @isa084 
If you plan on submitting your PR, I will be able to provide feedback.
In your screenshot I see that no `tohost_address` was passed to `Variane_testharness`. 
You can pass it like this `+tohost_addr=$(shell $$RISCV/bin/$(CV_SW_PREFIX)nm -B $(elf) | grep -w tohost | cut -d' ' -f1) \` but instead of the test path, use the pk binary. 
This should avoid timeout and give the correct test return value.
Also, I was wondering if the pk compilation was automatic and transparent to the user in your flow implementation ?

Thank you for your contribution.

isa084, 2025-02-17:
Hello @valentinThomazic ,
My PR is currently on hold as I am awaiting permission from my employer to make open source contributions to the CVA6 repository.

There are two possibilities ahead : 
(1) I will get permission from my employer, after which I will submit the PR.
(2) Someone else may take up my suggestions and create their own PR. In this way, progress is not stalled by my limitations.

Regarding, the compilation of pk : I followed the same steps that you had previously provided and there weren't any issues. I also started working on a script (similar to install-verilator.sh or install-spike.sh) to automatically install pk. It will make the installation of pk automatic and transparent (much to your liking).

I would suggest that we wait for a couple of weeks. As soon as I have clarity, I will inform you.

Thanks,
Isaar

isa084, 2025-03-09:
Update : I am still waiting for permission from my employer regarding contributions to this repository. I will keep you informed. Fingers crossed.

- Isaar

isa084, 2025-04-02:
Hi @valentinThomazic @JeanRochCoulon ,

A quick heads up. 

I am finalizing a pull request that introduces pk installation and veri-testharness simulation with pk.  Will send it across hopefully in 1-2 days.

For now, I have successfully tested it with cv64a6_imafdc_sv39.
I also tried the same flow with cv32a65x but the simulation stalls indefinitely and times out with 'AXI response errors'. I will share details and seek your advice on the compilation/simulation flags to make it functional.

The changes are transparent and non-invasive. Just set DV_SIMULATORS="veri-testharness-pk" and run it.
Those who don't want it, don't need to know or care.

valentinThomazic, 2025-04-02:
Hello @isa084, 
This is great!
I think it would be okay (to confirm with @JeanRochCoulon) if it works with only the configuration you mentionned.
However I think it is important that you add a test script so your work can be tested (you can look at one of the smoke tests scripts for instance).
You should know that I will be out of office starting next week for a long period (~2 months). 
If you manage to submit the PR this week I will be able to provide feedback.