openhwfoundation/cva6 #2432: [BUG] RVFI Trace (c.unimp when ecall)

MarioOpenHWGroup, 2024-08-02:
### Is there an existing CVA6 bug for this?

- [X] I have searched the existing bug issues

### Bug Description

## Environment

Hashes:
cva6: 6269f72b633a46c11abeb5054ea2cb28eb7ebb20
cvv: e06bd577f5db816015ec16a901e6e8f39cf7d640

## How to reproduce it

1. Download the binary attached
2. Place the binary in the `verif/sim/out*/directed_asm_tests/
3. Bypass the binary compilation on verif/sim/cva6.py 
4. Run smoke tests.

or 

Run the following command without downloading the binary
Command:
```
source verif/sim/setup-env.sh
SPIKE_TANDEM=1 NUM_JOBS=12 UVM_VERBOSITY=UVM_MEDIUM DV_SIMULATORS=vcs-testharness bash verif/regress/smoke-tests.sh
```

Binary compiled: [I-ADD-01.zip](https://github.com/user-attachments/files/16469855/I-ADD-01.zip)
DISCLAIMER: In some environments the binary produced does not have this ecall and it has a direct write to tohost

## Explanation

In the trace of the test `rv32i-I-ADD-01` on the configuration `cv32a6_imac_sv32`, when the `ecall` in the pc `0x000000008000031a` the tracer produces another entry with a `c.unimp`

rv32i-I-ADD-01.cv32a6_imac_sv32.log:

```
415 core   0: 0x0000000080000314 (0x0ff0000f) fence   iorw,iorw
416 3 0x0000000080000314 (0x0ff0000f)
417 core   0: 0x0000000080000318 (0x00004185) c.li    gp, 1
418 3 0x0000000080000318 (0x4185) x 3 0x00000001
419 core   0: 0x000000008000031a (0x00000073) ecall           <--- HERE
420 3 0x000000008000031a (0x00000073)
421 core   0: 0x000000008000031e (0x00000000) c.unimp        <---- HERE
422 3 0x000000008000031e (0x0000)
423 core   0: 0x0000000080000004 (0x34202f73) csrrs   t5, mcause, zero
424 3 0x0000000080000004 (0x34202f73) x30 0x0000000b
```

JeanRochCoulon, 2024-08-02:
@cathales do you have some comment on it ?

cathales, 2024-08-02:
https://github.com/openhwgroup/cva6/blob/0c60bc6e3dccd61753aeb2176fc1d586b283578b/core/cva6_rvfi.sv#L278C1-L281C62

Looks like `ecall` makes `rvfi` validate the instruction on both ports, without considering that only the first one is the call.

MarioOpenHWGroup, 2024-08-02:
I have done a implementation to get the ex_i from the scoreboard entry and I am seeing both of scoreboard entries with the ecall exception enabled.

MarioOpenHWGroup, 2024-08-23:
Solved, fix already in master branch

closed 2024-08-23 by MarioOpenHWGroup