openhwfoundation/cva6 #2433: Synthesis for CVA6 and ARA

Tanishqgithub, 2024-08-06:
I am currently working on synthesizing the ARA vector unit integrated with the CVA6 processor using Cadence Genus [ARA](https://github.com/pulp-platform/ara) While attempting this, I selected the ARA+CVA6 module as the top module and mapped all the files from a top-down approach. However, I have encountered numerous errors during synthesis, leading me to believe that my approach may be incorrect.

Given your expertise in this domain, I would greatly appreciate any guidance or suggestions you could provide to help me resolve these issues. Specifically, I am looking for advice on the following:
1. The correct methodology for synthesizing the ARA vector unit with the CVA6 processor in Cadence Genus.
2. Any specific considerations or configurations that need to be set to ensure successful synthesis.
3. Common pitfalls or errors that I should be aware of and how to address them.

If not possible in CADENCE GENUS can you also guide me through the tools like synopsis DC or Vivado

JeanRochCoulon, 2024-08-06:
This is the github issue dedicated to CVA6, and not to the integration of ARA with CVA6. I suggest you to open an issue in ARA repository and close this one.

Tanishqgithub, 2024-08-07:
I know this issue is slightly not related to this repository 

But as the ARA system is designed on CVA6 i thought some insights or guidance will help me achieve synthesis.
Is there any possibilities that I receive help regarding this

closed 2024-08-15 by Tanishqgithub