openhwfoundation/cva6 #2437: Missing cva6_config_pkg and CVA6Cfg Files Required for Synthesis

Vinay0203, 2024-08-08:
### Is there an existing CVA6 bug for this?

- [X] I have searched the existing bug issues

### Bug Description

During the synthesis process, it was discovered that the files cva6_config_pkg and CVA6Cfg are missing from the repository. These files are crucial for the configuration and successful synthesis. Their absence is currently preventing the synthesis from completing as expected. 
It would be greatly appreciated if you could assist in locating these files. If they are genuinely missing, please provide the necessary files to resolve this issue.

Gchauvon, 2024-08-12:
Hello @Vinay0203,

You might have forgotten to set the `TARGET` variable. CVA6 configuration is set into all files names `*_config_pkg.sv` in `core/include`. CVA6Cfg is parameter derived from this.

ibooga, 2024-08-17:
I'm having a similar problem after pulling the most recent version of the project. When running some of the CSR tests with the target set to cv32a60x I receive the error:

Cannot find file containing module: /home/cai/cache_project/cva6/core/include/cv32a60x_config_pkg.sv

closed 2024-08-23 by Vinay0203