openhwfoundation/cva6 #2442: Questions about cva6_config_pkg

hhhsiang, 2024-08-13:
### Is there an existing CVA6 bug for this?

- [X] I have searched the existing bug issues

### Bug Description

When running the smoke-tests.sh, I noticed it employs diverse "--target" options for each test, such as cv64a6_imafdc_sv39, cv32a65x, and cv64a6_mmu.

one of test in smoke-tsets.sh：
> python3 cva6.py --testlist=../tests/testlist_riscv-tests-cv64a6_imafdc_sv39-p.yaml --test rv64ui-p-addi --iss_yaml cva6.yaml --target cv64a6_imafdc_sv39 --iss=$DV_SIMULATORS $DV_OPTS

My question is：
1. Is "--target" used to select a specific cva6_config_pkg?
2. If the answer of question one is yes, which file is used to explicitly define the needed cva6_config_pkg in /core/include? For example, if the target is cv64a6_imafdc_sv39, where to define the corresponding cva6_config_pkg is "cv64a6_imafdc_sv39_config_pkg.sv"?
3. I think the "sv39" in cv64a6_imafdc_sv39 means that the virtual address is used, so the mmu is instantiated. Why we need "cv64a6_mmu"? What is the different between cv64a6_imafdc_sv39 and cv64a6_mmu?
4. Is there document to explain the user scenarios for every cva6_config_pkg?

JeanRochCoulon, 2024-08-13:
Thank you for your interest in CVA6. Github Issues is not the right place to ask for questions. Please use the Github Discussion which is dedicated to it.

closed 2024-08-13 by JeanRochCoulon

Moschn, 2024-08-13:
Sorry to chime in, but I believe this might be a relevant issue. It would be nice to document the main targeted configurations (or make it easier to find). Yes, this is "just" a documentation issue, but an issue nonetheless. 

I took it upon myself to try to answer this here from my limited understanding:

1. Yes, `--target` can be used to select the configuration.
2. Check the folder `core/include` for a file that matches the configuration that you are looking for. Note that the filenames might change in the future (and might no longer match with the target). You could also for example check the bender configuration for the exact file that is used: https://github.com/openhwgroup/cva6/blob/9b576c1200a69f7f3d92d07a18fe0b878f979a24/Bender.yml#L26-L56
3. This is where I am not 100% certain. I believe `cv64a_mmu` is just a shorter handle for a 64bit configuration with MMU. For example, `cv64a_mmu` should be equivalent to `cv64a6_imafdc_sv39`. 
4. There is some initial documentation to describe the major flavors: https://docs.openhwgroup.org/projects/cva6-user-manual/02_cva6_requirements/cva6_requirements_specification.html#initial-release. As far as I am aware, there is no further documentation on the individual configurations. The letters in the configuration (`imafdc`) come from the official risc-v specification.

hhhsiang, 2024-08-14:
@Moschn, thanks for your response! I also noticed the `Bender.yml`. However, I can't find the definition for `cv64a6_mmu `that matches `cv64a6_mmu_config_pkg.sv` in /core/include/, so I'm unsure if `Bender.yml` is the right file for defining the corresponding cva6_config_pkg.

Moschn, 2024-08-14:
`Bender.yml` is __not__ the official file to try to find the corresponding file. As far as I know, there is currently no official way to figure this out without diving into cva6.py. You can use `bender.yml` as a pointer though. 

Once again, I do not think `cv64a6_mmu` exists under this name in `core/include`. `cva6.py` seems to indicate that it is equivalent to  `cv64a6_imafdc_sv39` and thus uses its config file.