openhwfoundation/cva6 #2453: CVA6 Synthesis in Cadence Genus problem regarding structure cmo_req_t and cmo_rsp_t

Tanishqgithub, 2024-08-19:
I am doing module wise synthesis for CVA6 and then the whole ```cva6.sv ``` synthesis 

When i was trying to do synthesis of the module ``` cva6_hpdcache_subsystem ```  i encountered a missing structure used in this module and i could not find this structure in github 

The use of this is here in ```cva6_hpdcache_cmo_if_adapter.sv ``` : 
```
/*
 *  Copyright 2023 CEA*
 *  *Commissariat a l'Energie Atomique et aux Energies Alternatives (CEA)
 *
 *  SPDX-License-Identifier: Apache-2.0 WITH SHL-2.1
 *
 *  Licensed under the Solderpad Hardware License v 2.1 (the “License”); you
 *  may not use this file except in compliance with the License, or, at your
 *  option, the Apache License version 2.0. You may obtain a copy of the
 *  License at
 *
 *  https://solderpad.org/licenses/SHL-2.1/
 *
 *  Unless required by applicable law or agreed to in writing, any work
 *  distributed under the License is distributed on an “AS IS” BASIS, WITHOUT
 *  WARRANTIES OR CONDITIONS OF ANY KIND, either express or implied. See the
 *  License for the specific language governing permissions and limitations
 *  under the License.
 */
/*
 *  Authors       : Cesar Fuguet
 *  Creation Date : April, 2021
 *  Description   : Interface adapter for the CMO interface of the CVA6 core
 *  History       :
 */
module cva6_hpdcache_cmo_if_adapter
import hpdcache_pkg::*;

//  Parameters
//  {{{
#(
    parameter type cmo_req_t = logic,
    parameter type cmo_rsp_t = logic
)
//  }}}

//  Ports
//  {{{
(
  //  Clock and active-low reset pins
  input  logic                            clk_i,
  input  logic                            rst_ni,

  //  Port ID
  input  hpdcache_pkg::hpdcache_req_sid_t dcache_req_sid_i,

  //  Request/response ports from/to the CVA6 core
  input  cmo_req_t                        cva6_cmo_req_i,
  output cmo_rsp_t                        cva6_cmo_resp_o,

  //  Request port to the L1 Dcache
  output logic                            dcache_req_valid_o,
  input  logic                            dcache_req_ready_i,
  output hpdcache_pkg::hpdcache_req_t     dcache_req_o,

  //  Response port from the L1 Dcache
  input  logic                            dcache_rsp_valid_i,
  input  hpdcache_pkg::hpdcache_rsp_t     dcache_rsp_i
);
//  }}}

  //  Internal nets and registers
  //  {{{
  enum {
    FORWARD_IDLE,
    FORWARD_CMO,
    FORWARD_CMO_ACK
  } forward_state_q, forward_state_d;

  logic forward_cmo;
  hpdcache_pkg::hpdcache_req_t dcache_req_cmo;
  logic [ariane_pkg::TRANS_ID_BITS-1:0] cmo_tid_q, cmo_tid_d;
  logic cmo_ack;
  logic stall;
  //  }}}

  //  Request forwarding
  //  {{{
  always_comb
  begin : req_forward_comb
    forward_state_d = forward_state_q;
    forward_cmo     = 1'b0;
    cmo_tid_d       = cmo_tid_q;
    cmo_ack         = 1'b0;
    stall           = 1'b0;

    case (forward_state_q)
      FORWARD_IDLE: begin
        if (cva6_cmo_req_i.req) begin
          stall       = ~dcache_req_ready_i;
          forward_cmo = 1'b1;
          cmo_tid_d   = cva6_cmo_req_i.trans_id;
          if (!dcache_req_ready_i) begin
            forward_state_d = FORWARD_CMO;
          end else begin
            forward_state_d = FORWARD_CMO_ACK;
          end
        end
      end

      FORWARD_CMO: begin
        stall       = ~dcache_req_ready_i;
        forward_cmo = 1'b1;
        if (dcache_req_ready_i) begin
          forward_state_d = FORWARD_CMO_ACK;
        end
      end

      FORWARD_CMO_ACK: begin
        stall   = 1'b1;
        cmo_ack = 1'b1;
        if (cva6_cmo_req_i.req) begin
          stall       = ~dcache_req_ready_i;
          forward_cmo = 1'b1;
          cmo_tid_d   = cva6_cmo_req_i.trans_id;
          if (!dcache_req_ready_i) begin
            forward_state_d = FORWARD_CMO;
          end else begin
            forward_state_d = FORWARD_CMO_ACK;
          end
        end else begin
          forward_state_d = FORWARD_IDLE;
        end
      end
    endcase
  end

  always_ff @(posedge clk_i or negedge rst_ni)
  begin : forward_ff
    if (!rst_ni) begin
      forward_state_q <= FORWARD_IDLE;
      cmo_tid_q <= '0;
    end else begin
      forward_state_q <= forward_state_d;
      cmo_tid_q <= cmo_tid_d;
    end
  end

  //  CMO request
  //  {{{
  always_comb
  begin : cmo_req
    dcache_req_cmo.addr        = hpdcache_req_addr_t'(cva6_cmo_req_i.address);
    dcache_req_cmo.need_rsp    = 1'b0;
    dcache_req_cmo.uncacheable = 1'b0;
    dcache_req_cmo.sid         = dcache_req_sid_i;
    dcache_req_cmo.tid         = cva6_cmo_req_i.trans_id;
    dcache_req_cmo.wdata       = '0;
    dcache_req_cmo.be          = '0;
    dcache_req_cmo.op          = HPDCACHE_REQ_CMO;
    dcache_req_cmo.size        = '0;
    case (cva6_cmo_req_i.cmo_op)
      ariane_pkg::CMO_CLEAN,
      ariane_pkg::CMO_FLUSH,
      ariane_pkg::CMO_ZERO: begin
        // FIXME
      end  
      ariane_pkg::CMO_INVAL: begin   
        dcache_req_cmo.size = HPDCACHE_REQ_CMO_INVAL_NLINE;
      end
     ariane_pkg::CMO_PREFETCH_R,
     ariane_pkg::CMO_PREFETCH_W: begin 
        dcache_req_cmo.size = HPDCACHE_REQ_CMO_PREFETCH;
      end
      ariane_pkg::CMO_CLEAN_ALL,
      ariane_pkg::CMO_FLUSH_ALL: begin
      end 
      ariane_pkg::CMO_INVAL_ALL: begin  
        dcache_req_cmo.size = HPDCACHE_REQ_CMO_INVAL_ALL;
      end
    endcase
  end
  //  }}}

  assign dcache_req_valid_o        = forward_cmo,
         dcache_req_o              = dcache_req_cmo,
         cva6_cmo_resp_o.req_ready = ~stall;
  //  }}}

  //  Response forwarding
  //  {{{
  assign cva6_cmo_resp_o.ack       = cmo_ack,
         cva6_cmo_resp_o.trans_id  = cmo_tid_q;
  //  }}}

endmodule
```
and along with this the signals ```ariane_pkg::CMO_CLEAN, ariane_pkg::CMO_FLUSH, ariane_pkg::CMO_ZERO,  ariane_pkg::CMO_INVAL, ariane_pkg::CMO_PREFETCH_R, ariane_pkg::CMO_PREFETCH_W,   ariane_pkg::CMO_CLEAN_ALL,
ariane_pkg::CMO_FLUSH_ALL, ariane_pkg::CMO_INVAL_ALL```  are not defined the package ``` ariane_pkg ```

Please help me in locating these signals and the structures ``` cmo_req_t ; cmo_rsp_t ```

cfuguet, 2024-08-19:
Hello @Tanishqgithub,

The file you are synthesizing is not currently part of the CVA6. It is a component I made a time ago to enable the use of CMO requests from the CVA6 to the HPDcache. I sent a patch a time ago to @niwis to add all the necessary changes to support CMOs in the CVA6. Is this what you are using ?

cfuguet, 2024-08-19:
Or maybe you are using a old version of CVA6 ? I may have at some point commited that file but I removed it later. If you are in that case (old version of CVA6), either you can update it or remove the CMO adapter from the file list to synthesize. It is not required

Tanishqgithub, 2024-08-20:
@cfuguet 

I found this while synthesizing ARA vector unit in ``` ara/deps/cva6/cva6.sv ``` in : 
```

    // -------------------
  // Cache Subsystem
  // -------------------

  // Acc dispatcher and store buffer share a dcache request port.
  // Store buffer always has priority access over acc dipsatcher.
  dcache_req_i_t [NumPorts-1:0] dcache_req_to_cache;
  dcache_req_o_t [NumPorts-1:0] dcache_req_from_cache;

  // D$ request
  assign dcache_req_to_cache[0] = dcache_req_ports_ex_cache[0];
  assign dcache_req_to_cache[1] = dcache_req_ports_ex_cache[1];
  assign dcache_req_to_cache[2] = dcache_req_ports_acc_cache[0];
  assign dcache_req_to_cache[3] = dcache_req_ports_ex_cache[2].data_req ? dcache_req_ports_ex_cache [2] :
                                                                          dcache_req_ports_acc_cache[1];

  // D$ response
  assign dcache_req_ports_cache_ex[0] = dcache_req_from_cache[0];
  assign dcache_req_ports_cache_ex[1] = dcache_req_from_cache[1];
  assign dcache_req_ports_cache_acc[0] = dcache_req_from_cache[2];
  always_comb begin : gen_dcache_req_store_data_gnt
    dcache_req_ports_cache_ex[2]  = dcache_req_from_cache[3];
    dcache_req_ports_cache_acc[1] = dcache_req_from_cache[3];

    // Set gnt signal
    dcache_req_ports_cache_ex[2].data_gnt &= dcache_req_ports_ex_cache[2].data_req;
    dcache_req_ports_cache_acc[1].data_gnt &= !dcache_req_ports_ex_cache[2].data_req;
  end

  if (DCACHE_TYPE == int'(config_pkg::WT)) begin : gen_cache_wt
    // this is a cache subsystem that is compatible with OpenPiton
    wt_cache_subsystem #(
        .CVA6Cfg   (CVA6ExtendCfg),
        .NumPorts  (NumPorts),
        .noc_req_t (noc_req_t),
        .noc_resp_t(noc_resp_t)
    ) i_cache_subsystem (
        // to D$
        .clk_i             (clk_i),
        .rst_ni            (rst_uarch_n),
        .busy_o            (busy_cache_ctrl),
        .stall_i           (stall_ctrl_cache),
        .init_ni           (init_ctrl_cache_n),
        // I$
        .icache_en_i       (icache_en_csr),
        .icache_flush_i    (icache_flush_ctrl_cache),
        .icache_miss_o     (icache_miss_cache_perf),
        .icache_areq_i     (icache_areq_ex_cache),
        .icache_areq_o     (icache_areq_cache_ex),
        .icache_dreq_i     (icache_dreq_if_cache),
        .icache_dreq_o     (icache_dreq_cache_if),
        // D$
        .dcache_enable_i   (dcache_en_csr_nbdcache),
        .dcache_flush_i    (dcache_flush_ctrl_cache),
        .dcache_flush_ack_o(dcache_flush_ack_cache_ctrl),
        // to commit stage
        .dcache_amo_req_i  (amo_req),
        .dcache_amo_resp_o (amo_resp),
        // from PTW, Load Unit  and Store Unit
        .dcache_miss_o     (dcache_miss_cache_perf),
        .miss_vld_bits_o   (miss_vld_bits),
        .dcache_req_ports_i(dcache_req_to_cache),
        .dcache_req_ports_o(dcache_req_from_cache),
        // write buffer status
        .wbuffer_empty_o   (dcache_commit_wbuffer_empty),
[... 92 lines cut ...]
        .icache_dreq_o     (icache_dreq_cache_if),
        // D$
        .dcache_enable_i   (dcache_en_csr_nbdcache),
        .dcache_flush_i    (dcache_flush_ctrl_cache),
        .dcache_flush_ack_o(dcache_flush_ack_cache_ctrl),
        // to commit stage
        .amo_req_i         (amo_req),
        .amo_resp_o        (amo_resp),
        .dcache_miss_o     (dcache_miss_cache_perf),
        // this is statically set to 1 as the std_cache does not have a wbuffer
        .wbuffer_empty_o   (dcache_commit_wbuffer_empty),
        // from PTW, Load Unit  and Store Unit
        .dcache_req_ports_i(dcache_req_to_cache),
        .dcache_req_ports_o(dcache_req_from_cache),
        // memory side
        .axi_req_o         (noc_req_o),
        .axi_resp_i        (noc_resp_i)
    );
    assign dcache_commit_wbuffer_not_ni = 1'b1;
    assign inval_ready                  = 1'b1;
  end

```
although i am now working out with ``` wt_cache_subsystem ``` while eliminating the use of ``` cva6_hpdcache_subsystem ``` 
But i would wish to synthesis ``` cva6_hpdcache_subsystem ``` as well

cfuguet, 2024-08-20:
@Tanishqgithub, yes, you can synthesize with the HPDcache. What Git repository and branch are you using ? I do not find the ara/deps/cva6/cva6.sv file in the CVA6 official repository (https://github.com/openhwgroup/cva6).

DerekTsai, 2024-08-20:
@cfuguet Hi, sir. Can I get the latest version of cva6_hpdcache_cmo_if_adapter? thanks.

cfuguet, 2024-08-20:
> @cfuguet Hi, sir. Can I get the latest version of cva6_hpdcache_cmo_if_adapter? thanks.

It is the one above.

Tanishqgithub, 2024-08-21:
@cfuguet 
I got this file cloned after i installed the ARA vector processor in [ARA](https://github.com/pulp-platform/ara) 
Although the ``` deps ``` folder is not directly available in github, i think it gets initiated after the Bender links the cva6 to ARA 

Can you please provide me with the Structure defined for ```cmo_req_t ; cmo_rsp_t  ``` and along with the signals ``` ariane_pkg::CMO_CLEAN, ariane_pkg::CMO_FLUSH, ariane_pkg::CMO_ZERO,  ariane_pkg::CMO_INVAL, ariane_pkg::CMO_PREFETCH_R, ariane_pkg::CMO_PREFETCH_W,   ariane_pkg::CMO_CLEAN_ALL, ariane_pkg::CMO_FLUSH_ALL, ariane_pkg::CMO_INVAL_ALL ``` which are not in ``` ariane_pkg```

cfuguet, 2024-08-21:
Ok. I think the easiest thing to do is to remove the cmo_adapter from the Bender manifest files. You won't have any issue because it is not used by default.

Another thing is that I'm not sure the HPDcache subsystem is currently supported by the ARA platform. I would recommend you to write to the PULP team and ask them. Maybe @niwis have an answer for that ?

Tanishqgithub, 2024-08-22:
@cfuguet @niwis 
Thanks for the reply