openhwfoundation/cva6 #2462: Hello world standalone simulation error (True/2)

Yidaen, 2024-08-25:
Hello everyone,

I am new to the platform and trying to understand how it works. Therefore, I installed a new Linux 22.04 and followed the instructions in the repository to set up the toolchain, Spike, and Verilator.

Afterward, I ran the "hello world" simulation using the commands provided in the repository and encountered the following error:

`ERROR return code: True/2, cmd:make veri-testharness target=cv32a60x variant=rv32imac_zba_zbb_zbs_zbc_zicsr_zifencei elf=/home/new_era/culsans/project/cva6/verif/sim/out_2024-08-25/directed_c_tests/hello_world.o path_var=/home/new_era/culsans/project/cva6/ tool_path=/home/new_era/culsans/project/cva6/tools/spike/bin isscomp_opts="" issrun_opts="+debug_disable=1 +ntb_random_seed=1" isspostrun_opts="0x0000000080000000" log=/home/new_era/culsans/project/cva6/verif/sim/out_2024-08-25/veri-testharness_sim/hello_world.cv32a60x.log &> /home/new_era/culsans/project/cva6/verif/sim/out_2024-08-25/veri-testharness_sim/hello_world.cv32a60x.log.iss
`

When I checked the error description in the hello_world.cv32a60x.log.iss file, I found:

`%Error: Cannot find file containing module: /home/new_era/culsans/project/cva6/core/include/cv32a60x_config_pkg.sv`

Even though the file exists, it could not be found.

When I checked the directory, the location information for the relevant file showed a different path:

`corev_apu/axi_node//home/new_era/culsans/project/cva6/core/include/cv32a60x_config_pkg.sv
`
What should I do about this error? Thank you for your help.

JeanRochCoulon, 2024-08-25:
cv32a60x is deprecated. You need to select another target cv32a65x for instance. Thank you to close the issue if solved.

bibhuti-k, 2024-08-28:
i tried to run using same command target=cv32a65x and i am getting the same error:
ERROR return code: True/2, cmd:make vcs-testharness target=cv32a65x variant=rv32imc_zba_zbb_zbs_zbc_zicsr_zifencei elf=/home/test/Desktop/cva6/verif/sim/out_2024-08-28/directed_c_tests/hello_world.o path_var=/home/test/Desktop/cva6/ tool_path=/home/test/Desktop/cva6/tools/spike/bin isscomp_opts="" issrun_opts="+debug_disable=1 +ntb_random_seed=1" isspostrun_opts="0x0000000080000000" log=/home/test/Desktop/cva6/verif/sim/out_2024-08-28/vcs-testharness_sim/hello_world.cv32a65x.log &> /home/test/Desktop/cva6/verif/sim/out_2024-08-28/vcs-testharness_sim/hello_world.cv32a65x.log.iss

when I tried to find error message in log file  hello_world.cv32a65x.log.iss there is no error message. What can be done now

bibhuti-k, 2024-08-28:
got this error message while running using target cv32a65x 
vcs -full64   -ignore initializer_driver_checks -timescale=1ns/1ns -ntb_opts uvm-1.2 work.ariane_tb -error="IWNF" \
 +notimingcheck
/bin/sh: 2: vlogan: not found
make[1]: *** [Makefile:324: vcs_build] Error 127
make[1]: Leaving directory '/home/test/Desktop/cva6'
make: *** [Makefile:397: vcs-testharness] Error 2
This error is present in log file. How can this error be resolved.

valentinThomazic, 2024-08-29:
It seems you are trying to run the simulation with vcs but have an issue with your path. Did you intend to run it with verilator instead like in your first post ?

closed 2024-09-29 by JeanRochCoulon