openhwfoundation/cva6 #2470: [BUG] PMP benchmark/test is broken

Moschn, 2024-08-27:
### Is there an existing CVA6 bug for this?

- [X] I have searched the existing bug issues

### Bug Description

A couple of years ago, CVA6 did manage to successfully run the PMP benchmark. The PMP benchmark tries to set a lot of PMP rules and tests some corner cases, so it is a good stresstest for an implementation. 

I tested the current CVA6 version and it no longer passes the PMP benchmark/test. Not sure where this regression originated. 

We should definitely fix this (and then add the PMP benchmark to the CI). I no longer have the necessary setup to quickly see what is going on, so it would be appreciated if someone else could take a look.

This could also help the verification effort (#2457).

Steps to reproduce:
```
# to make it run the PMP benchmark directly, comment out all other benchmarks in verif/regress/benchmark.sh
DV_SIMULATORS=veri-testharness bash verif/regress/benchmark.sh
```

Example output:
```
Makefile:14: must set CVA6_REPO_DIR to point at the root of CVA6 sources and CVA6_TB_DIR to point here -- doing it for you...
make -C /scratch/ariane/cva6/ verilate verilator="verilator --no-timing" target=cv64a6_imafdc_sv39 defines=
make[1]: Entering directory '/scratch/ariane/cva6'
Makefile:150: XCELIUM_HOME not set which is necessary for compiling DPIs when using XCELIUM
[Verilator] Building Model
verilator --no-timing --no-timing verilator_config.vlt -f core/Flist.cva6 core/cva6_rvfi.sv /scratch/ariane/cva6/corev_apu/tb/ariane_axi_pkg.sv /scratch/ariane/cva6/corev_apu/tb/axi_intf.sv /scratch/ariane/cva6/corev_apu/register_interface/src/reg_intf.sv /scratch/ariane/cva6/corev_apu/tb/ariane_soc_pkg.sv /scratch/ariane/cva6/corev_apu/riscv-dbg/src/dm_pkg.sv /scratch/ariane/cva6/corev_apu/tb/ariane_axi_soc_pkg.sv /scratch/ariane/cva6/core/cva6_rvfi.sv /scratch/ariane/cva6/corev_apu/src/ariane.sv /scratch/ariane/cva6/corev_apu/bootrom/bootrom.sv /scratch/ariane/cva6/corev_apu/clint/axi_lite_interface.sv /scratch/ariane/cva6/corev_apu/clint/clint.sv /scratch/ariane/cva6/corev_apu/fpga/src/axi2apb/src/axi2apb_64_32.sv /scratch/ariane/cva6/corev_apu/fpga/src/axi2apb/src/axi2apb.sv /scratch/ariane/cva6/corev_apu/fpga/src/axi2apb/src/axi2apb_wrap.sv /scratch/ariane/cva6/corev_apu/fpga/src/apb_timer/apb_timer.sv /scratch/ariane/cva6/corev_apu/fpga/src/apb_timer/timer.sv /scratch/ariane/cva6/corev_apu/fpga/src/axi_slice/src/axi_ar_buffer.sv /scratch/ariane/cva6/corev_apu/fpga/src/axi_slice/src/axi_aw_buffer.sv /scratch/ariane/cva6/corev_apu/fpga/src/axi_slice/src/axi_b_buffer.sv /scratch/ariane/cva6/corev_apu/fpga/src/axi_slice/src/axi_r_buffer.sv /scratch/ariane/cva6/corev_apu/fpga/src/axi_slice/src/axi_single_slice.sv /scratch/ariane/cva6/corev_apu/fpga/src/axi_slice/src/axi_slice.sv /scratch/ariane/cva6/corev_apu/fpga/src/axi_slice/src/axi_slice_wrap.sv /scratch/ariane/cva6/corev_apu/fpga/src/axi_slice/src/axi_w_buffer.sv /scratch/ariane/cva6/corev_apu/src/axi_riscv_atomics/src/axi_res_tbl.sv /scratch/ariane/cva6/corev_apu/src/axi_riscv_atomics/src/axi_riscv_amos_alu.sv /scratch/ariane/cva6/corev_apu/src/axi_riscv_atomics/src/axi_riscv_amos.sv /scratch/ariane/cva6/corev_apu/src/axi_riscv_atomics/src/axi_riscv_atomics.sv /scratch/ariane/cva6/corev_apu/src/axi_riscv_atomics/src/axi_riscv_atomics_wrap.sv /scratch/ariane/cva6/corev_apu/src/axi_riscv_atomics/src/axi_riscv_lrsc.sv /scratch/ariane/cva6/corev_apu/src/axi_riscv_atomics/src/axi_riscv_lrsc_wrap.sv /scratch/ariane/cva6/corev_apu/axi_mem_if/src/axi2mem.sv /scratch/ariane/cva6/corev_apu/rv_plic/rtl/rv_plic_target.sv /scratch/ariane/cva6/corev_apu/rv_plic/rtl/rv_plic_gateway.sv /scratch/ariane/cva6/corev_apu/rv_plic/rtl/plic_regmap.sv /scratch/ariane/cva6/corev_apu/rv_plic/rtl/plic_top.sv /scratch/ariane/cva6/corev_apu/riscv-dbg/src/dmi_cdc.sv /scratch/ariane/cva6/corev_apu/riscv-dbg/src/dmi_jtag.sv /scratch/ariane/cva6/corev_apu/riscv-dbg/src/dmi_jtag_tap.sv /scratch/ariane/cva6/corev_apu/riscv-dbg/src/dm_csrs.sv /scratch/ariane/cva6/corev_apu/riscv-dbg/src/dm_mem.sv /scratch/ariane/cva6/corev_apu/riscv-dbg/src/dm_sba.sv /scratch/ariane/cva6/corev_apu/riscv-dbg/src/dm_top.sv /scratch/ariane/cva6/corev_apu/riscv-dbg/debug_rom/debug_rom.sv /scratch/ariane/cva6/corev_apu/register_interface/src/apb_to_reg.sv /scratch/ariane/cva6/vendor/pulp-platform/axi/src/axi_multicut.sv /scratch/ariane/cva6/vendor/pulp-platform/common_cells/src/rstgen_bypass.sv /scratch/ariane/cva6/vendor/pulp-platform/common_cells/src/rstgen.sv /scratch/ariane/cva6/vendor/pulp-platform/common_cells/src/addr_decode.sv /scratch/ariane/cva6/vendor/pulp-platform/common_cells/src/stream_register.sv /scratch/ariane/cva6/vendor/pulp-platform/axi/src/axi_cut.sv /scratch/ariane/cva6/vendor/pulp-platform/axi/src/axi_join.sv /scratch/ariane/cva6/vendor/pulp-platform/axi/src/axi_delayer.sv /scratch/ariane/cva6/vendor/pulp-platform/axi/src/axi_to_axi_lite.sv /scratch/ariane/cva6/vendor/pulp-platform/axi/src/axi_id_prepend.sv /scratch/ariane/cva6/vendor/pulp-platform/axi/src/axi_atop_filter.sv /scratch/ariane/cva6/vendor/pulp-platform/axi/src/axi_err_slv.sv /scratch/ariane/cva6/vendor/pulp-platform/axi/src/axi_mux.sv /scratch/ariane/cva6/vendor/pulp-platform/axi/src/axi_demux.sv /scratch/ariane/cva6/vendor/pulp-platform/axi/src/axi_xbar.sv /scratch/ariane/cva6/vendor/pulp-platform/common_cells/src/cdc_2phase.sv /scratch/ariane/cva6/vendor/pulp-platform/common_cells/src/spill_register_flushable.sv /scratch/ariane/cva6/vendor/pulp-platform/common_cells/src/spill_register.sv /scratch/ariane/cva6/vendor/pulp-platform/common_cells/src/deprecated/fifo_v1.sv /scratch/ariane/cva6/vendor/pulp-platform/common_cells/src/deprecated/fifo_v2.sv /scratch/ariane/cva6/vendor/pulp-platform/common_cells/src/stream_delay.sv /scratch/ariane/cva6/vendor/pulp-platform/common_cells/src/lfsr_16bit.sv /scratch/ariane/cva6/vendor/pulp-platform/tech_cells_generic/src/deprecated/cluster_clk_cells.sv /scratch/ariane/cva6/vendor/pulp-platform/tech_cells_generic/src/deprecated/pulp_clk_cells.sv /scratch/ariane/cva6/vendor/pulp-platform/tech_cells_generic/src/rtl/tc_clk.sv /scratch/ariane/cva6/corev_apu/tb/ariane_testharness.sv /scratch/ariane/cva6/corev_apu/tb/ariane_peripherals.sv /scratch/ariane/cva6/corev_apu/tb/rvfi_tracer.sv /scratch/ariane/cva6/corev_apu/tb/common/uart.sv /scratch/ariane/cva6/corev_apu/tb/common/SimDTM.sv /scratch/ariane/cva6/corev_apu/tb/common/SimJTAG.sv +define+ corev_apu/tb/common/mock_uart.sv +incdir+corev_apu/axi_node  --unroll-count 256 -Wall -Werror-PINMISSING -Werror-IMPLICIT -Wno-fatal -Wno-PINCONNECTEMPTY -Wno-ASSIGNDLY -Wno-DECLFILENAME -Wno-UNUSED -Wno-UNOPTFLAT -Wno-BLKANDNBLK -Wno-style  -DPRELOAD=1     -LDFLAGS "-L/opt/riscv3/lib -L/scratch/ariane/cva6/tools/spike/lib -Wl,-rpath,/opt/riscv3/lib -Wl,-rpath,/scratch/ariane/cva6/tools/spike/lib -lfesvr -lriscv -ldisasm -lyaml-cpp  -lpthread " -CFLAGS "-I/include -I/include -I/scratch/ariane/cva6/tools/verilator-v5.008/share/verilator/include/vltstd -I/opt/riscv3/include -I/scratch/ariane/cva6/tools/spike/include -std=c++17 -I/scratch/ariane/cva6/corev_apu/tb/dpi -O3 -DVL_DEBUG -I/scratch/ariane/cva6/tools/spike"   --cc --vpi  +incdir+/scratch/ariane/cva6/vendor/pulp-platform/common_cells/include/  +incdir+/scratch/ariane/cva6/vendor/pulp-platform/axi/include/  +incdir+/scratch/ariane/cva6/corev_apu/register_interface/include/  +incdir+/scratch/ariane/cva6/corev_apu/tb/common/  +incdir+/scratch/ariane/cva6/vendor/pulp-platform/axi/include/  +incdir+/scratch/ariane/cva6/verif/core-v-verif/lib/uvm_agents/uvma_rvfi/  +incdir+/scratch/ariane/cva6/verif/core-v-verif/lib/uvm_components/uvmc_rvfi_reference_model/  +incdir+/scratch/ariane/cva6/verif/core-v-verif/lib/uvm_components/uvmc_rvfi_scoreboard/  +incdir+/scratch/ariane/cva6/verif/core-v-verif/lib/uvm_agents/uvma_core_cntrl/  +incdir+/scratch/ariane/cva6/verif/tb/core/  +incdir+/scratch/ariane/cva6/core/include/  +incdir+/scratch/ariane/cva6/tools/spike/include/disasm/ --top-module ariane_testharness --threads-dpi none --Mdir work-ver -O3 --exe corev_apu/tb/ariane_tb.cpp corev_apu/tb/dpi/SimDTM.cc corev_apu/tb/dpi/SimJTAG.cc corev_apu/tb/dpi/remote_bitbang.cc corev_apu/tb/dpi/msim_helper.cc
cd work-ver && make -j -f Variane_testharness.mk
make[2]: Entering directory '/scratch/ariane/cva6/work-ver'
make[2]: Nothing to be done for 'default'.
make[2]: Leaving directory '/scratch/ariane/cva6/work-ver'
make[1]: Leaving directory '/scratch/ariane/cva6'
[... cut ...] ri-testharness_sim/pmp.cv64a6_imafdc_sv39.log.yaml +core_name=cv64a6_imafdc_sv39
This emulator compiled with JTAG Remote Bitbang client. To enable, use +jtag_rbb_enable=1.
Listening on port 38261
/scratch/ariane/cva6/verif/sim/out_2024-08-27/directed_c_tests/pmp.o *** FAILED *** (tohost = 2) after 1326 cycles
*** [rvfi_tracer] INFO: Simulation terminated after       1314 cycles!

CPU time used: 13645.82 ms
Wall clock time passed: 13660.65 ms
make: *** [Makefile:412: veri-testharness] Error 2
```

hhhsiang, 2024-10-16:
Hi @Moschn ,I seem to find a error in pmp module when I run synthesis.
[image]

It said that the port number of pmp mismatch with the port in cv6_mmu module.
the signal `conf_addr_i` in pmp like this：
[image]

And the signal `pmpcfg_i` connect to `conf_addr_i` in cva_mmu module like this：
[image]

The number of port of `conf_addr_i` is `[NR_ENTRIES:0]`, but the number of port of `pmpcfg_i` is `[CVA6Cfg.NrPMPEntries-1:0]`.
The number of port is different. I think the number of port of `conf_addr_i` should be `[NR_ENTRIES-1:0]`.

What do you think?

Moschn, 2024-10-16:
Hi,

Thanks for the report. I believe you are correct. Would you mind fixing this and posting your first PR? I believe the fix should just change this line to `[NR_ENTRIES-1:0]`: https://github.com/openhwgroup/cva6/blob/dff627162b4337288cee6821e9138f1d8efc5caf/core/pmp/src/pmp.sv#L26