openhwfoundation/cva6 #2477: Issue With AXI_DATA_WIDTH parameter overiding

dvusingh, 2024-08-29:
Hi,

I am using the cv32a6_imac_sv32_config_pkg.sv config file to run the benchmark median test.
In that file when I tried to modify the AxiDataWidth from 64 bits to 128 bits **localparam CVA6ConfigAxiDataWidth = 128**. and ran the simulations.

I am getting the below Errors

**Error-[IRIPS] Illegal range in part select
home/cva6/core/cache_subsystem/wt_axi_adapter.sv, 173
  The range of the part select is illegal:
  AxiRdBlenDcache[($clog2(AxiNumWords) - 1):0]**

I saw the dependency of AxiDataWidth parameter with other parameters like ICACHE_LINE_WIDTH  & DCACHE_LINE_WIDTH  to calculate AxiNumWords.

 **localparam AxiNumWords = (CVA6Cfg.ICACHE_LINE_WIDTH/CVA6Cfg.AxiDataWidth) * (CVA6Cfg.ICACHE_LINE_WIDTH > 
                                                    CVA6Cfg.DCACHE_LINE_WIDTH)  +
                                          (CVA6Cfg.DCACHE_LINE_WIDTH/CVA6Cfg.AxiDataWidth) * (CVA6Cfg.ICACHE_LINE_WIDTH <= 
                                           CVA6Cfg.DCACHE_LINE_WIDTH) ;**

Is it correct to run the cva6 test with AxiDataWidth=128. Or what are the modification we need to do if we change AxiDataWidth.

It would be helpful if I get some clarification on this issue to proceed further.

Thanks

Gchauvon, 2024-09-18:
It looks like a wrong configuration of the cache. I believe you will have to check legal values in the code. It might cover a bug for the 128 bits configuration.
Once you have resolved the adapter issue, you should look into axi dedicated package also which can influence the platform

https://github.com/openhwgroup/cva6/blob/master/corev_apu/tb/ariane_axi_pkg.sv
https://github.com/openhwgroup/cva6/blob/master/corev_apu/tb/ariane_axi_soc_pkg.sv

closed 2024-10-24 by github-actions[bot]

khandelwaltanuj, 2025-07-25:
Hi @JeanRochCoulon @ASintzoff 

I have came acros the same issue in my simulation where 128 bit tests are not passing. I am geting illegal instruction error. 
It feels in my case as if memory model being used is not behaving correctly for 128 bits axidatawidth. 

Regards,

Tanuj Khandelwal