openhwfoundation/cva6 #2480: [TASK] zephyr RTOS support on Ariane SDK

WorldofJARcraft, 2024-08-30:
### Is there an existing CVA6 task for this?

- [X] I have searched the existing task issues

### Task Description

Hello all, I wanted to let you know that I am working on adding support for the CVA6 CPU in the [zephyr RTOS](https://github.com/zephyrproject-rtos/zephyr).

### Required Changes

No hardware changes are needed, but I would like to add a link to the board directory, instructions etc. in the CVA6 README once zephyr accepts my pull requests.

### Current Status

I have created three draft pull requests in zephyr: 
1. https://github.com/zephyrproject-rtos/zephyr/pull/77792 adds support for the CVA6 SoC on the GenesysII board that is contained in this repository.
2. https://github.com/zephyrproject-rtos/zephyr/pull/77734 adds support for my forked CVA6 SoC with Xilinx AXI Ethernet.
3. https://github.com/zephyrproject-rtos/zephyr/pull/77792 adds support for running zephyr in the cva6 test harness. I have added examples that demonstrate how to indicate success/failure from within zephyr to the testbench.

I have tested 1. and 3. extensively, and I will test 2. with a recent build of the CVA6 project ASAP.

### Risks

The zephyr configuration might go out of sync if devices are *removed* from the SoC or registers or other hardware configurations change, rendering the zephyr configuration unusable.
In order to combat this, I can document the CVA6 tag that I last tested with zephyr on the zephyr side.

### Prerequisites

None as this is only documentation.

### KPI (KEY Performance Indicators)

None

### Description of Done

Once zephyr has accepted my PRs, a link to the boards in zephyr should be added to the CVA6 README.

### Associated PRs

See above

WorldofJARcraft, 2024-08-30:
I have also sent PRs with drivers for the Xilinx AXI ethernet system to zephyr. As my board configurations depend on these drivers, and the PRs for the drivers are not merged yet, it might take a while for the PRs I have linked above to be merged.

JeanRochCoulon, 2024-09-19:
Hello,   Your proposal seems interesting. Maybe @jquevremont can help.

jquevremont, 2024-09-27:
Hi @WorldofJARcraft. When the PR is merged, can you ping me so that I can complete the upcoming RESOURCES documentt with the links (https://github.com/openhwgroup/cva6/pull/2514)?

WorldofJARcraft, 2024-09-27:
I will let you know when the PR in zephyr is merged.

jquevremont, 2024-10-28:
Hi @WorldofJARcraft, any progress on the (external) PR merging?

WorldofJARcraft, 2024-10-28:
My PR for zephyr also includes an option for supporting the Xilinx AXI Ethernet subsystem. I have also contributed PRs for these drivers to zephyr, which have not been merged yet. 
I plan to merge board support after merging the drivers, so this might take a few more weeks.

jquevremont, 2024-10-28:
Thanks. Please keep us updated.

jquevremont, 2024-11-28:
Hi @WorldofJARcraft. Are you also pinged by the robot?

WorldofJARcraft, 2024-11-28:
Yes.
I am still waiting for reviews on my current Pull Request in zephyr: https://github.com/zephyrproject-rtos/zephyr/pull/77732

PR #2949 (2025-07-24): Document how to load and run zephyr RTOS
Starting with the upcoming release 4.2.0 (and the current main branch), the zephyr RTOS has support for running on the 32/64 bit configurations of cva6 on the Genesys2 board.
This PR adds instructions for building and loading zephyr for these configurations to the README.
It also adds a script that can be used to load zephyr from an SD card.
I also have automated scripts that setup the build dependencies for zephyr and build a sample application - I can add them to the PR and add a corresponding GitHub action if you are interested.
This closes #2480.
commit 05bb5a48 (2025-07-24): Document how to load and run zephyr on FPGA (#2949)

Starting with the upcoming release 4.2.0 (and the current main branch), the zephyr RTOS has support for running on the 32/64 bit configurations of cva6 on the Genesys2 board.
This PR adds instructions for building and loading zephyr for these configurations to the README.
It also adds a script that can be used to load zephyr from an SD card.
I also have automated scripts that setup the build dependencies for zephyr and build a sample application - I can add them to the PR and add a corresponding GitHub action if you are interested.
This clos [... cut]
closed 2025-07-24 by JeanRochCoulon