openhwfoundation/cva6 #2487: CVA6 synthesis

hhhsiang, 2024-08-31:
### Is there an existing CVA6 bug for this?

- [X] I have searched the existing bug issues

### Bug Description

I'm looking to synthesize the CVA6 core and perform gate-level simulation. According to the guidelines, the following command is recommended for synthesis: `make -C pd/synth cva6_synth FOUNDRY_PATH=/your/techno/basepath/ TECH_NAME=yourTechnoName TARGET_LIBRARY_FILES="yourLib1.db\ yourLib2.db" PERIOD=10 NAND2_AREA=650 TARGET=cv64a6_imafdc_sv39 ADDITIONAL_SEARCH_PATH="others/libs/paths/one\ others/libs/paths/two"` However, upon inspecting the Makefile in `/cva6/pd/synth/`, I noticed a `# Deprecated` comment above the `cva6_synth` entry. Does this mean that the `cva6_synth` command is outdated? Is there a different command recommended for synthesizing the CVA6 now?

ibooga, 2024-09-17:
I'm curious about this as well. I would like to perform synthesis and layout, ideally with the cv32a60x

ibooga, 2024-09-18:
Great, thanks for the advice.

An additional question, if I may.

Are there any examples of a (verified?) cva6 that have been brought close to tape out that provide GDS2 files or something similar? It would be nice to have a frame of reference while synthesizing and verifying my variations.

Thank you for your help, cheers

Gchauvon, 2024-09-19:
There have been cva6 that have been put to silicon. However I do not have anything to share to you.

hhhsiang, 2024-09-19:
Hi @Gchauvon, I noticed your comment about no longer updating or supporting the "old" cva6_synth target. Does that mean you will not synthesize cva6 core and run gate simulation anymore? Given that the cva6 core is a 64-bit CPU, distinct from the 32-bit cv32a65x, does this indicate that the cv64a6 core will not be maintained in the future? Thanks for clarifying!

hhhsiang, 2024-09-20:
@Gchauvon, I'm curious why the cv64a6 core is maintained but you don't run CI on the cv64a6_imafdc_sv39 target, and there is no synthesis or gate-level simulation. What if the RTL design is correct but issues arise at the gate level? How do you ensure the design's functionality without these steps? Could you provide more insights into the reasoning behind this decision and its potential impact on the project? Thank you!

Gchauvon, 2024-09-20:
There are non-regression tests on CI with the cv64a6_imafdc_sv39 target. This is to be sure we do not break this target at simulation level. We do not guarantee that the design is correct at gate level. 
In Thales, we don't have the ressource to verify all posible configuration so we have decided to verify only cv32a65x. Other target may be used and maintained by other people.

JeanRochCoulon, 2025-01-02:
@hhhsiang We are cleaning the "Github Issues". May I ask you to close this issue if it is the case ?

closed 2025-01-19 by hhhsiang