openhwfoundation/cva6 #2489: CVA6 Pulp Platform issue

dvusingh, 2024-09-03:
Hi,

I am working on the cva6 verification.
There I found that some of the rtl files present under the **cva6/vendor/pulp_platform/axi & cva6/vendor/pulp_platform/common_cells** are old in comparison to the **github pulp-platform** area shown below:
https://github.com/pulp-platform/axi
https://github.com/pulp-platform/common_cells/tree/master/src

If I will replace the older cva6/vendor/pulp_platform with updated files present in github pulp-platform and run the cva6 simulations,
So, does its affect the functionality of the cva6 core.

I want some clarification on this to procced effectively to understand the cva6 design implementation. 
Any information would be really appreciated. 

Thanks

niwis, 2024-09-03:
Hello @dvusingh, pulp platform projects generally follow [semantic versioning](https://semver.org/spec/v2.0.0.html), meaning that minor releases (e.g. `v1.2.0 -> v1.3.0`) should be fully backwards-compatible. However, [pulp-platform/axi](https://github.com/pulp-platform/axi) is still pre-release (`v0.x.y`), so functional/interface changes are possible. Upgrading dependencies is usually not a big deal, but it's always a good idea to run regression tests afterward, as they should catch any functional changes.

dvusingh, 2024-09-10:
Hi,

I ran benchmark test case, I don't see any issues with updated pulp_platform files.

Thanks

closed 2024-10-28 by github-actions[bot]