openhwfoundation/cva6 #2495: Looking at possibility of running testbenches without using a compiler

natitomato, 2024-09-06:
Hello, 
I am new to cva6, and currently trying to look at the verification aspect of the core. I do not want to use a compiler (which i believe is being used in the verification environment).
I had a look into `verif/tb/core` directory and I'm not sure if the stand-alone testbench uses a compiler (with my tiny bit of research I don't think it does).
Are there any workarounds or testbenches instantiating the core which i can just run with Cadence or Synopsys without using the compiler?

Any help would be appreciated.

Thank you,
Samarth B S

Gchauvon, 2024-09-18:
Unfortunately all of our testbench require a program to be loaded to memory.

samarthbs27, 2024-09-18:
Thanks for the reply!

However I have a question, I had posted a query here: [Issue#2498](https://github.com/openhwgroup/cva6/issues/2498) and from this I understand that [ariane_gate_tb](https://github.com/openhwgroup/cva6/blob/master/corev_apu/tb/ariane_gate_tb.sv) is the top module instantiating the core, and it is loading the binary to memory.
If i generate a binary by myself and make the top module point to it, could I just run this top module by itself with any simulation tool?

Gchauvon, 2024-09-19:
I believe it should work fine.

samarthbs27, 2024-09-19:
Thanks for your input. I will reopen the issue if I get another query related to this in the future.

closed 2024-09-19 by samarthbs27