openhwfoundation/cva6 #2496: [BUG] mstatus.fs is always OFF even with fpu instantiated

FriedDede, 2024-09-06:
### Is there an existing CVA6 bug for this?

- [X] I have searched the existing bug issues

### Bug Description

Hi, i'm trying to run bare metal software on a cva6 core, but i'm encountering a bug with fp instructions. Every fp instruction getd marked as illegal because mstatus.fs is always == riscv::Off.

This signal seems to come from the csr, but evey assign to this signal is either 
`mstatus_d.fs = riscv::Dirty;`
or
`mstatus_d.fs = riscv::Off;` 
I don't see it beign assigned Initial or Clean as i would expect anywhere.

Is this a bug? Do i have to unlock the fpu via software by manually writing mstatus (if it is even possible)?

Thank you for your support.
Andrea

FS is always == Off
[image]

So the instruction `fmv.w.x	fa5,zero` is not recognized as an fpu instruction, and marked illegal down the line
[image]

This is my cva6 config:
package cva6_config_pkg;

  localparam CVA6ConfigXlen = 64;

  localparam CVA6ConfigRVF = 1;
  localparam CVA6ConfigF16En = 1;
  localparam CVA6ConfigF16AltEn = 1;
  localparam CVA6ConfigF8En = 1;
  localparam CVA6ConfigFVecEn = 1;

  localparam CVA6ConfigCvxifEn = 0;
  localparam CVA6ConfigCExtEn = 1;
  localparam CVA6ConfigZcbExtEn = 1;
  localparam CVA6ConfigZcmpExtEn = 0;
  localparam CVA6ConfigAExtEn = 1;
  localparam CVA6ConfigHExtEn = 0;  // always disabled
  localparam CVA6ConfigBExtEn = 0;
  localparam CVA6ConfigVExtEn = 0;
  localparam CVA6ConfigRVZiCond = 0;

  localparam CVA6ConfigAxiIdWidth = 4;
  localparam CVA6ConfigAxiAddrWidth = 64;
  localparam CVA6ConfigAxiDataWidth = 64;
  localparam CVA6ConfigFetchUserEn = 0;
  localparam CVA6ConfigFetchUserWidth = CVA6ConfigXlen;
  localparam CVA6ConfigDataUserEn = 0;
  localparam CVA6ConfigDataUserWidth = CVA6ConfigXlen;

  localparam CVA6ConfigIcacheByteSize = 16384;
  localparam CVA6ConfigIcacheSetAssoc = 4;
  localparam CVA6ConfigIcacheLineWidth = 128;
  localparam CVA6ConfigDcacheByteSize = 32768;
  localparam CVA6ConfigDcacheSetAssoc = 8;
  localparam CVA6ConfigDcacheLineWidth = 128;

  localparam CVA6ConfigDcacheIdWidth = 1;
  localparam CVA6ConfigMemTidWidth = 2;

  localparam CVA6ConfigWtDcacheWbufDepth = 8;

  localparam CVA6ConfigNrScoreboardEntries = 8;

  localparam CVA6ConfigFpgaEn = 1;

  localparam CVA6ConfigNrLoadPipeRegs = 1;
  localparam CVA6ConfigNrStorePipeRegs = 0;
  localparam CVA6ConfigNrLoadBufEntries = 2;

  localparam CVA6ConfigRASDepth = 2;
  localparam CVA6ConfigBTBEntries = 32;
  localparam CVA6ConfigBHTEntries = 128;

  localparam CVA6ConfigTvalEn = 1;

  localparam CVA6ConfigNrPMPEntries = 8;

  localparam CVA6ConfigPerfCounterEn = 1;

  localparam config_pkg::cache_type_t CVA6ConfigDcacheType = config_pkg::WT;

  localparam CVA6ConfigMmuPresent = 0;

  localparam CVA6ConfigRvfiTrace = 1;

Here is my fpu imp:

PipeRegs:   '{
                '{
                unsigned'(LAT_COMP_FP32),
                unsigned'(LAT_COMP_FP64),
                unsigned'(LAT_COMP_FP16),
                unsigned'(LAT_COMP_FP8),
                unsigned'(LAT_COMP_FP16ALT),
                unsigned'(LAT_COMP_FP8ALT)
                },
                '{default: unsigned'(LAT_DIVSQRT)}, // DIVSQRT
                '{default: unsigned'(LAT_NONCOMP)}, // NONCOMP
                '{default: unsigned'(LAT_CONV)},     // CONV
                '{default: unsigned'(LAT_DOTP)}     // CONV
              },  // ADDMUL,
  UnitTypes:  '{'{default: PARALLEL}, // ADDMUL
                '{default: MERGED}, // DIVSQRT
                '{default: PARALLEL}, // NONCOMP
                '{default: MERGED},   // CONV
                '{default: DISABLED}},  // DOTP
  PipeConfig: DISTRIBUTED

closed 2024-09-09 by FriedDede

MikeOpenHWGroup, 2024-09-19:
Hi @FriedDede, thanks for your interest in CVA6.  I see you closed this issue - were you able to resolve it yourself?

FriedDede, 2024-09-20:
> Hi @FriedDede, thanks for your interest in CVA6. I see you closed this issue - were you able to resolve it yourself?

Hi @MikeOpenHWGroup , yes I was able to resolve myself the issue. The FPU needs to be explicitly enabled writing mstatus.fs.

I've added those lines to my bootrom and now is working great.
`// initialize fpu
  li t0, 1 << 13  
  csrw mstatus, t0`

Thanks you for the support anyway.