openhwfoundation/cva6 #2498: [BUG] <title> Makefile for stand-alone testbench is failing

samarthbs27, 2024-09-10:
### Is there an existing CVA6 bug for this?

- [X] I have searched the existing bug issues

### Bug Description

Hello,

I'm having trouble running the **core only testbench** in `cva6/verif/tb/core/` Makefile with Cadence Xcelium.
I am aware that this is an experimental stand-alone as specified in the README but any kind of clarification or remark would be helpful.

Starting here:
```
[vlsilab@c2s-06 cva6]$ cd verif/tb/core/
[vlsilab@c2s-06 core]$ make xrun_run
Makefile:29: must set CVA6_REPO_DIR to point at the root of CVA6 sources and CVA6_TB_DIR to point here -- doing it for you...
[XRUN] Building Model
mkdir -p xrun_results
cd xrun_results && \
xrun -sv -64bit -disable_sem2009 -access +rwc -timescale 1ns/1ps  -nowarn UEXPSC -uvm -uvmhome CDNS-1.2-ML -l xrun_compl.log -f /home/vlsilab/openhw/cva6/core/Flist.cv64a6_imafdc_sv39 -f /home/vlsilab/openhw/cva6/verif/tb/core//Flist.cva6_tb -top cva6_core_only_tb -elaborate
xrun: 20.09-s012: (c) Copyright 1995-2021 Cadence Design Systems, Inc.
   "-f /home/vlsilab/openhw/cva6/core/Flist.cv64a6_imafdc_sv39" referenced on the xrun command line
xrun: *F,BDARGF: command line argument file '/home/vlsilab/openhw/cva6/core/Flist.cv64a6_imafdc_sv39' could not be opened for reading.
make: *** [xrun_comp] Error 2
```

The Makefile points to a manifest file in cva6/core looking for `Flist.cv64a6_imafdc_sv39` which does not exist.
I found `Flist.cva6` and `Flist.cva6_gate`.
I modified the **target** in the Makefile to `cva6` (Since i believe cva6_gate is for synthesis)
```
Original: target     ?= cv64a6_imafdc_sv39

Modified: target     ?= cva6
```

After the above change, I get the following error:
```
[vlsilab@c2s-06 core]$ make xrun_run
Makefile:29: must set CVA6_REPO_DIR to point at the root of CVA6 sources and CVA6_TB_DIR to point here -- doing it for you...
[XRUN] Building Model
mkdir -p xrun_results
cd xrun_results && \
xrun -sv -64bit -disable_sem2009 -access +rwc -timescale 1ns/1ps  -nowarn UEXPSC -uvm -uvmhome CDNS-1.2-ML -l xrun_compl.log -f /home/vlsilab/openhw/cva6/core/Flist.cva6 -f /home/vlsilab/openhw/cva6/verif/tb/core//Flist.cva6_tb -top cva6_core_only_tb -elaborate
xrun: 20.09-s012: (c) Copyright 1995-2021 Cadence Design Systems, Inc.
   "-F /rtl/hpdcache.Flist" used in file /home/vlsilab/openhw/cva6/core/Flist.cva6
   "-f /home/vlsilab/openhw/cva6/core/Flist.cva6" referenced on the xrun command line
xrun: *F,BDARGF: command line argument file '/rtl/hpdcache.Flist' could not be opened for reading.
make: *** [xrun_comp] Error 2
```

I added these lines to define **HPDCACHE_DIR** in the Makefile:
```
ifndef HPDCACHE_DIR
export HPDCACHE_DIR   = $(CVA6_REPO_DIR)/core/cache_subsystem/hpdcache/rtl)
endif
```

Output:
```
[vlsilab@c2s-06 core]$ make xrun_run
Makefile:29: must set CVA6_REPO_DIR to point at the root of CVA6 sources and CVA6_TB_DIR to point here -- doing it for you...
[XRUN] Building Model
mkdir -p xrun_results
cd xrun_results && \
xrun -sv -64bit -disable_sem2009 -access +rwc -timescale 1ns/1ps  -nowarn UEXPSC -uvm -uvmhome CDNS-1.2-ML -l xrun_compl.log -f /home/vlsilab/openhw/cva6/core/Flist.cva6 -f /home/vlsilab/openhw/cva6/verif/tb/core//Flist.cva6_tb -top cva6_core_only_tb -elaborate
TOOL:	xrun(64)	20.09-s012: Started on Sep 10, 2024 at 19:27:31 IST
xrun(64): 20.09-s012: (c) Copyright 1995-2021 Cadence Design Systems, Inc.
xrun: *E,FILEMIS: Cannot find the provided file /home/vlsilab/openhw/cva6/core/include/_config_pkg.sv.
TOOL:	xrun(64)	20.09-s012: Exiting on Sep 10, 2024 at 19:27:31 IST  (total: 00:00:00)
make: *** [xrun_comp] Error 1
```

So i added the following lines to the Makefile too to point to the target config file.
```
ifndef TARGET_CFG
export TARGET_CFG   = cv64a6_imafdc_sv39
endif
```

Which gave me this, and this is where I don't really know where to look or what to look for:
```
[vlsilab@c2s-06 core]$ make xrun_run
Makefile:29: must set CVA6_REPO_DIR to point at the root of CVA6 sources and CVA6_TB_DIR to point here -- doing it for you...
[XRUN] Building Model
mkdir -p xrun_results
cd xrun_results && \
xrun -sv -64bit -disable_sem2009 -access +rwc -timescale 1ns/1ps  -nowarn UEXPSC -uvm -uvmhome CDNS-1.2-ML -l xrun_compl.log -f /home/vlsilab/openhw/cva6/core/Flist.cva6 -f /home/vlsilab/openhw/cva6/verif/tb/core//Flist.cva6_tb -top cva6_core_only_tb -elaborate
TOOL:	xrun(64)	20.09-s012: Started on Sep 10, 2024 at 19:34:35 IST
xrun(64): 20.09-s012: (c) Copyright 1995-2021 Cadence Design Systems, Inc.
Compiling UVM packages (uvm_pkg.sv cdns_uvm_pkg.sv) using uvmhome location /home/install/XCELIUM2009/tools/methodology/UVM/CDNS-1.2-ML
file: /home/vlsilab/openhw/cva6/corev_apu/tb/rvfi_tracer.sv
            read_symbol("tohost", TOHOST_ADDR);
                      |
xmvlog: *W,FUNTSK (/home/vlsilab/openhw/cva6/corev_apu/tb/rvfi_tracer.sv,52|22): function called as a task without void'().
file: /home/vlsilab/openhw/cva6/verif/tb/core//tb_components/axi_master_connect.sv
xmvlog: *W,SPDUSD: Include directory /home/vlsilab/openhw/cva6/vendor/pulp-platform/common_cells/src/ given but not used.
xmvlog: *W,SPDUSD: Include directory /home/vlsilab/openhw/cva6/vendor/pulp-platform/axi/include/ given but not used.
	Total errors/warnings found outside modules and primitives:
		errors: 0, warnings: 2
xmelab: *E,NOUNIT: Unable to find a unit named 'cva6_core_only_tb' in the libraries.
xrun: *E,ELBERR: Error during elaboration (status 1), exiting.
TOOL:	xrun(64)	20.09-s012: Exiting on Sep 10, 2024 at 19:34:37 IST  (total: 00:00:02)
make: *** [xrun_comp] Error 1
```
I can't seem to find a `cva6_core_only_tb` or anything of that sort.
Is the `cva6_core_only_tb` supposed to be an empty tb that just let's the core run for a few thousand cycles?

Thank you.

Gchauvon, 2024-09-18:
This looks like you trying to use old and deprecated code. Our "core only" testbench today is our UVM testbench which is supported with synopsys VCS and is dedicated to the cv32a65x configuration verification. 

You might want to port the "gate simulation testbench " [ariane_gate_tb](https://github.com/openhwgroup/cva6/blob/master/corev_apu/tb/ariane_gate_tb.sv) to Xcelium. It is a simpler testbench than [testharness](https://github.com/openhwgroup/cva6/blob/master/corev_apu/tb/ariane_tb.sv). It is supported with Synopsys today but most of the flow is common to the `testharness tb` which i believe is supported by Xcelium

Gchauvon, 2024-09-19:
It is correct but you can also you [ariane_gate_tb](https://github.com/openhwgroup/cva6/blob/master/corev_apu/tb/ariane_gate_tb.sv) for simulation.

samarthbs27, 2024-09-19:
Ok thanks!

closed 2024-09-19 by samarthbs27