openhwfoundation/cva6 #2500: [BUG] : mip.MSIP and mie.MSIE are read-only zeros

AyoubJalali, 2024-09-12:
### Is there an existing CVA6 bug for this?

- [X] I have searched the existing bug issues

### Bug Description

According to [CV32A65X Spec](https://docs.openhwgroup.org/projects/cva6-user-manual/04_cv32a65x/riscv/priv.html#_machine_interrupt_mip_and_mie_registers), `As the system has only one hart then mip.MSIP and mie.MSIE are read-only zeros.`

So there's a bug on RTL, it allow modifying mip.MSIP and mie.MSIE

JeanRochCoulon, 2024-09-13:
This issue has not been detected up-to-now, the reason is maybe linked to the fact that Spike get the same RTL behavior allowing to modify mip.MSIP and mie.MSIE. So, there' maybe a bug in Spike.

AyoubJalali, 2024-09-13:
> This issue has not been detected up-to-now, the reason is maybe linked to the fact that Spike get the same RTL behavior allowing to modify mip.MSIP and mie.MSIE. So, there' maybe a bug in Spike.

I opend an issue on core-v-verif related to the same bug in spike

JeanRochCoulon, 2024-10-02:
This issue is present in both Spike and RTL. It was hidden by SpikeTandem

zchamski, 2024-10-16:
The Spike side was fixed in https://github.com/openhwgroup/cva6/pull/2523.  The remaining action is to update the RTL and add an appropriate RTL configuration parameter, as pointed out by @AyoubJalali.

valentinThomazic, 2024-11-18:
The CSR embedded regression test is failing because of this.  
Mismatch example : 
```
mismatch_description: 'CSR 304  Mismatch   [REF]: 0x880              [CORE]: 0x888             '
mismatches:
- 0: null
  core:
    insn: 0000000030419073
    insn_disasm: csrrw   zero, mie, gp
    mode: 3
    pc_rdata: 00000000800036ac
    pc_wdata: 0
    rd1_addr: 0
    rd1_rdata: 0
    rs1_addr: 3
    rs1_rdata: 0000000000000008
    rs2_addr: 4
    rs2_rdata: 196
    trap: 0
  reference_model:
    insn: 0000000030419073
    insn_disasm: csrrw   zero, mie, gp
    mode: 3
    pc_rdata: 00000000800036ac
    pc_wdata: 0
    rd1_addr: 0
    rd1_rdata: 0
    rs1_addr: 3
    rs1_rdata: 0000000000000008
    rs2_addr: 4
    rs2_rdata: 0000000080000000
    trap: 0
```

This should be resolved as soon as the RTL is aligned on Spike

JeanRochCoulon, 2025-01-16:
#2711 closes the issue

closed 2025-01-16 by JeanRochCoulon