openhwfoundation/cva6 #2510: Lockstep co-simulation

jason23g, 2024-09-24:
Hello guys,

By reading the README files for the verification and simulations, I understand that the co-simulation of CVA6 is being achieved by comparing the log trace of one RTL simulators such as verilator or vcs with the log trace of the Spike simulator. However, I view in another closed issue that for CV32A65X there is the possibility for lockstep co-simulation between vcs and spike, but I can not find where this is happening in the repo. Could you clarify if it is implemented the lockstep co-simulation and where they are the corresponding scripts?

Thanks in advance for your time.

JeanRochCoulon, 2024-09-24:
The Spike Tandem is implemented, this is a co-simulation of RTL and Spike to check the RTL behaviour. This is implemented for cv32a65x and not already deployed on other CVA6 configurations. To enable it, you should set SPIKE_TANDEM=1 when executing a simulation with DV_SIMULATOR=vcs-uvm (or vcs-testhaness or veri-testharness).

jason23g, 2024-09-25:
Thanks for your answer, indeed set the parameters as you indicate and run a co-simulation of RTL and spike, however in the log info shows that the TANDEM YAML report is being incomplete is this something that should happen or not ? 

[image]

jason23g, 2024-09-27:
UPDATE !!

SETUP :  
 The setup of the whole system has been made following the README file of the repository. After system setup i set the env variable SPIKE_TANDEM = 1, UVM_VERBOSITY=UVM_MEDIUM and the DV_SIMULATOR= veri-testharness. Then i have created a custom script to run only tests for the configuration CVA32A65X. 

PROBLEM : 

The first strange thing that i observe is the message "Incomplete TANDEM YAML report". After a thorough inspection of the source code of cva6.py, the sv files used for the simulation and the generated log files with expansion .log.iss i observe that even if the final messages are saying *** SUCESS  *** , the function rvfi_gen_report in the file uvmc_rvfi_scoreboard_utils.sv is never being executed and so on it is never producing a yaml report. The reason for that is both the reference_model_t and t_core instance have al the fields for the whole simulation time equal to zero.

QUESTION : 
Is this a bug or did i miss anything in the process setting up  the repository.

[logs.zip](https://github.com/user-attachments/files/17164956/logs.zip)
[custom_test.zip](https://github.com/user-attachments/files/17164960/custom_test.zip)
[image]

JeanRochCoulon, 2024-10-22:
We meet difficulties to make functional Verilator in Tandem mode. Verification flow using VCS is more robust.

valentinThomazic, 2024-11-06:
Related to https://github.com/openhwgroup/cva6/issues/2526, please take a look at my answer there.

JeanRochCoulon, 2024-12-07:
This issue is open for a while without any activity. I close it. Please open a new one if needed.

closed 2024-12-07 by JeanRochCoulon