openhwfoundation/cva6 #2525: AXI-4 Protocol for AMO

Arhum-Ahmed, 2024-10-04:
Hi,

I was testing the AMO instructions. I observed that in the axi adapter, communication on AW and W channels are being carried out simultaneously (in the same cycle). I also checked the TB of pulp repo and found the same interface there. But, according to the AXI-4 protocol specification, first the data should be written on AW channel and in the next cycle in the W channel. So, when I changed the interface, according to the specification, the tests started to hang. I wanted to know that is this interface implementation specific to this AMO only, or am I missing something?

niwis, 2024-10-04:
Hi @Arhum-Ahmed, the AW and W channels are decoupled, and there are no requirements for ordering in AXI. See A3.4 of the AXI spec (issue K):
> The protocol does not define any other relationship between the channels.
>
> The lack of relationship means, for example, that the write data can appear at an interface before the write request for the transaction. This can occur if the write request channel contains more register stages than the write data channel. Similarly, the write data might appear in the same cycle as the request.

Where do you see the requirement for sequencing AW and W?

Arhum-Ahmed, 2024-10-04:
Actually the problem occurs when I change the AMO's TB transaction behavior from simultaneous communication on the AW and W channels to sequenced communication (first using the AW channel, then the W channel). After making this change, I execute a load-reserved/store-conditional (lr/sc) AMO instruction with AW.LOCK set to 1 (for the store-conditional operation). This change causes incorrect response from the B channel. What might be the issue for this response?

closed 2024-11-10 by github-actions[bot]