openhwfoundation/cva6 #2526: [BUG] Spike tandem simulation not working as expected with the Verilator

jason23g, 2024-10-04:
### Is there an existing CVA6 bug for this?

- [X] I have searched the existing bug issues

### Bug Description

Hello guys,

I am trying to simulate cva6 with spike tandem, however i observe an unexpected behavior when i use spike tandem with the veri-testharness option. More specifically, i have tried to simulate a very simple execution of an assembly file. During the spike tandem simulation i observe that even if the files .log (Verilator Simulation) and tandem.log (spike simulation) work as expected, initially the message info says that the TANDEM YAML report is incomplete meaning that it doesn't print all the necessary information required. Furthermore when i change UVM_VERBOSITY variable to  UVM_FULL value (i tried all other options too)  or i manually tried to print information about what the core has executed and what the reference model has executed instruction by instruction, the only thing that i get is the c.unimp instruction and all the values of the instruction equal to 0. So my question is even if the problem lies down to the support of UVM in Verilator and so on this is the expected behavior or i am missing something else.

Thanks in advance for your time.

Note : I have attached all the files that i use to test it as well as the log files.
[files.zip](https://github.com/user-attachments/files/17256970/files.zip)

JeanRochCoulon, 2024-10-09:
Naive question: are you in simulation timeout ?   This can lead to this kind of output message.

jason23g, 2024-10-09:
No i am not in simulation timeout, i have not changed any of the other parameters except the environmental variables needed for the spike tandem simulation and the elf program that i use it is really small around 10 assembly instructions.

hhhsiang, 2024-10-14:
hi @jason23g ,sorry to bother you. Does "spike tandem" means that we use spike as iss to compare the simulation with verilator?

jason23g, 2024-10-15:
Hello @hhhsiang,

Spike tandem means that the co-simulation between Spike ( An Instruction Set Simulator ) and Verilator (An RTL Simulator) is lockstep, it checks the result between Spike and Verilator instruction by instruction. Whereas in the normal co-simulation it is just checking the traces of the 2 simulations at the end of them.

JeanRochCoulon, 2025-01-02:
@jason23g We are cleaning the "Github Issues". I assume this issue is closed. May I ask you to close it ?

jason23g, 2025-01-05:
Yes

closed 2025-01-05 by jason23g