openhwfoundation/cva6 #2531: Questions about Replacing SRAM Model for cache system and Issues for Synthesis

hhhsiang, 2024-10-05:
### Is there an existing CVA6 bug for this?

- [X] I have searched the existing bug issues

### Bug Description

Hi all,

I am planning to synthesize the entire system, including the CVA6 core, AXI bus, and DRAM. However, I noticed that the SRAM macro (`tc_sram.sv`), which is instantiated in the cache subsystem, is a behavioral model intended for RTL simulation. To proceed with synthesis, I believe I need to replace the `tc_sram.sv` with my own SRAM module. Is that correct?

Additionally, are there any other important issues or considerations I should be aware of before synthesizing the system?

Any advice or suggestions would be greatly appreciated. Thanks in advance!

niwis, 2024-10-07:
Hi @hhhsiang, yes, the intended usage of the [generic tech cells](https://github.com/pulp-platform/tech_cells_generic) is to choose a definition according to your target. The definition in this repo is indeed only a behavioural model intended for RTL simulation and should be replaced by a macro for your target technology when synthesised (i.e. by changing the definition of `tc_sram` such that it wraps your macro). I hope this helps.
By the way, I don't think that 'Bug' is the appropriate type for this issue.

niwis, 2024-10-08:
Sure, I think that would work.

JeanRochCoulon, 2024-10-09:
Please close the issue if it can be

closed 2024-10-11 by hhhsiang