openhwfoundation/cva6 #2534: [BUG] <title>VERILATOR_INSTALL_DIR is not set proprely in setup-env.sh

Mabdalshakour, 2024-10-07:
### Is there an existing CVA6 bug for this?

- [X] I have searched the existing bug issues

### Bug Description

In setup-env.sh line 45, VERILATOR_INSTALL_DIR is set to "$ROOT_PROJECT"/tools/verilator which doesn't take the VERILATOR_HASH into account like what install-verilator.sh does in line 40, where it sets it to $ROOT_PROJECT/tools/verilator-$VERILATOR_HASH. That makes verilator unreachable except for running smoke-tests.sh. You can't source setup-env.sh as a standalone.

valentinThomazic, 2024-10-28:
Maybe I am failing to see the issue but I don't see it on the current latest commit (https://github.com/openhwgroup/cva6/commit/ab2283c075db94499d791a0eef66da2449cc2c1b) since the `verif/regress/install-verilator.sh` script will install the binary in `tools/verilator/bin` which is the default path  in the `verif/sim/setup-env.sh` script.
However, I think either the `README` or `verif/sim/setup-env.sh` need update because I cannot run standalone simulations only with `README` instructions.

JeanRochCoulon, 2025-01-02:
@Mabdalshakour We are cleaning the "Github Issues". I assume this issue is closed. May I ask you to close it ?

valentinThomazic, 2025-02-17:
Given that no answer has been given for 4 months, I close this issue.
Feel free to reopen it or create another one if needed.

closed 2025-02-17 by valentinThomazic