openhwfoundation/cva6 #2535: Error generating fpga bitsream

daksith, 2024-10-07:
### Is there an existing CVA6 bug for this?

- [X] I have searched the existing bug issues

### Bug Description

I get the following output on the command line while trying to generate the bitstream on vivado 2024.1

ERROR: [Vivado 12-13638] Failed runs(s) : 'xlnx_mig_7_ddr3_synth_1'
wait_on_runs: Time (s): cpu = 00:00:57 ; elapsed = 00:01:16 . Memory (MB): peak = 2289.176 ; gain = 0.000 ; free physical = 1395 ; free virtual = 1795
ERROR: [Common 17-39] 'wait_on_runs' failed due to earlier errors.

    while executing
"wait_on_run ${ipName}_synth_1"
    (file "tcl/run.tcl" line 20)
INFO: [Common 17-206] Exiting Vivado at Mon Oct  7 12:53:36 2024...

make[2]: *** [../common.mk:2: all] Error 1
make[2]: Leaving directory '/home/daksith/PULP/tes_fpga/cva6/corev_apu/fpga/xilinx/xlnx_mig_7_ddr3'
make[1]: *** [Makefile:34: work-fpga/xlnx_mig_7_ddr3.xci] Error 2
make[1]: Leaving directory '/home/daksith/PULP/tes_fpga/cva6/corev_apu/fpga'
make: *** [Makefile:748: fpga] Error 2

Yanze66, 2024-10-08:
I encounter the same issue. Waiting for any solution.

daksith, 2024-10-11:
Worked on Vivado 2020.2 for me.

Gchauvon, 2024-10-15:
In CI, we are using Vivado 2018.2 because our version is linked to the year we bought the FPGA board.

JeanRochCoulon, 2025-01-02:
@daksith We are cleaning the "Github Issues". I assume this issue is closed. May I ask you to close it ?