openhwfoundation/cva6 #2539: Error in Makefile

Hasan0712, 2024-10-09:
Hello,
During the build process of our project (specifically when working with the Ariane repository), we encountered a linker error that prevented the creation of the executable.
[image]
verilator version 4.110
ariane repo:https://github.com/lowRISC/ariane.git

JeanRochCoulon, 2024-10-09:
Here is the support of cva6 processor hosted in cva6 repository. The compilation/simulation flow of lowRISCV/ariane is not suported here. Can you kindly post this issue in dedicated repository and close this issue?

Hasan0712, 2024-10-10:
Okay i will sir but i am also facing the same issue in this ariane too can
you help me out there

Hasan0712, 2024-10-11:
> Your branch is out of date. Follow the cva6 README, in that case Verilator
> will be v5.x and the CVA6 is updated. This should work. If you meeet an
> issue, open a new one.

https://github.com/openhwgroup/cva6/issues/2542
The current issue i am facing