openhwfoundation/cva6 #2544: [BUG] error when synthesis

hhhsiang, 2024-10-14:
### Is there an existing CVA6 bug for this?

- [X] I have searched the existing bug issues

### Bug Description

When I synthesize cv64a6 core with `cv64a6_imafdc_sv39_config_pkg.sv` with Design Compiler version-2023.12, there are lots of error occurred, and all of them like the following：
[image]

> Error: ../..//core/scoreboard.sv:303: Constant expression too complex. (ELAB-937)

It's all about "Constant expression too complex. (ELAB-937)". I check the systemverilog code where the error happened. All of the errors seem to be caused by the parameter of cva6 config pkg.

[image]

When the parameters of cva6_config_pkg are used to determine whether a circuit or a module will be instantiated, the "Constant expression too complex. (ELAB-937)" error occurred. Lots of modules are not synthesized. How can I fixed it?

Any advice or suggestions would be greatly appreciated. Thanks in advance!

JeanRochCoulon, 2024-10-14:
Hello @hhhsiang     The synthesis is run on cv32a65x, no experience on 64 bit configuration.
Do you experience the same type of error on cv32a65x configuration ?

hhhsiang, 2024-10-15:
Hi @JeanRochCoulon , as far as i know, the cv64a6 was synthesized in the past, so there are scripts in Makefile under the `cva6/pd/synth` directory. Because in the old commit cv64a6 is the synthesis target in the makefile, I'm confused why did you say no experience on 64 bit configuration?

I also tyr to synthesize cv32a65x, with `make -C pd/synth cva6_synth FOUNDRY_PATH=/your/techno/basepath/ TECH_NAME=yourTechnoName TARGET_LIBRARY_FILES="yourLib1.db\ yourLib2.db" PERIOD=10 NAND2_AREA=650 TARGET=cv64a6_imafdc_sv39 ADDITIONAL_SEARCH_PATH="others/libs/paths/one\ others/libs/paths/two"`, but there is error occurred like the following：
[image]
 It seems that `$(SYNTH_FLOW_NAME)` parameter is not set properly.

JeanRochCoulon, 2024-10-15:
Yes, you are right, I personally setup maybe 3 years ago the scripts to synthetize 32 and 64 bits configurations. But we currently use different scripts to do it. That is why these scripts are no more used.

hhhsiang, 2024-10-15:
Understood. Could you provide the script you use for synthesizing the 64-bit configuration? I think it is difficult to construct a script to successfully synthesize cv64a6 core because there are so many dependeUnderstood. Could you provide the script you use for synthesizing the 64-bit configuration? I think it is difficult to construct a script to successfully synthesize cv64a6 core because there are so many dependent file need to be include and preprocess.nt file need to be include and preprocess.

JeanRochCoulon, 2024-10-15:
Unfortunately, physical implementation depends on techno which we cannot disclose it.

JeanRochCoulon, 2025-01-02:
@hhhsiang We are cleaning the "Github Issues". I assume this issue is closed. May I ask you to close it ?

hhhsiang, 2025-01-19:
@ergulfaruk , you need to make sure that the parameters which is used to instantiate modules has no "dot", such as "CVA6Cfg.FpPresent" or "CVA6Cfg.NrCommitPorts" in my exmaples. You need to change them into constant.