openhwfoundation/cva6 #2547: Spike Tandem - Configuration Support

jason23g, 2024-10-15:
Hello guys,

Currently the spike tandem is working only with the configuration of CV32A65X, i want to ask what i need to do in the spike.yaml file  if i want to implement another configuration for the spike tandem. For instance  cv32a6_imac_sv32 or cv64a6_imafdc_sv39 except changing the isa to the appropriate one, what values need to have the fields marchid_override_mask, marchid_override_value, misa_write_mask and all the others.

Thanks in advance for your time.

JeanRochCoulon, 2024-10-15:
Hello @jason23g 
Do you have time to attend the OpenHW meeting ?  It would be the good place to ask for such information. It is on THursday afternoon, ask for invitation to @ASintzoff if needed.

JeanRochCoulon, 2024-10-22:
Hello @jason23g 
A documentation presenting Spike Tandem has just been pushed on CVA6 repository.
To create a new Spike configuration, you need to create a new spike.yaml with your configuration.
I close this issue, and feel free to open a new one for another issue.

closed 2024-10-22 by JeanRochCoulon