openhwfoundation/cva6 #2554: [BUG][online docs] Broken links in top-level README.md

zchamski, 2024-10-17:
### Is there an existing CVA6 bug for this?

- [X] I have searched the existing bug issues

### Bug Description

The top-level `README.md` contains broken links:

* [Build Status](https://camo.githubusercontent.com/9ff2dee66f99859e54fc00ae12337b369257f9a946c2cccf79961d5ae46aee6c/68747470733a2f2f72697363762d63692e70616765732e7468616c65732d696e7669612e66722f64617368626f6172642f) results in a "Content length exceeded" error
* [CVA6 dashboard](https://github.com/openhwgroup/cva6/blob/master/util/toolchain-builder/README.md#Prerequisites) points to the wrong location ("Prerequisites" section of the README file instead of the actual dashboard).

valentinThomazic, 2024-11-06:
Hey @zchamski , I think this can be closed :)

zchamski, 2024-11-06:
The result looks great, closing issue😀

closed 2024-11-06 by zchamski