openhwfoundation/cva6 #2556: [BUG] RVFI agent & tandem mode simulation performance degradation

AnouarZajni, 2024-10-17:
### Is there an existing CVA6 bug for this?

- [X] I have searched the existing bug issues

### Bug Description

Hi,
RVFI agent + spike Tandem is degrading considerably cva6 simulation time due to monitoring & scoreboarding CSRs in CSR_MAX_SIZE (4096) instead of CSR supported size (Actual cv32a65x number of CSRs is 187). uvma_rvfi and spike tandem should be parametrized to support only needed number of CSRs

Profiling coremark test with nocoverage, no logger and rvfi_disable_csr_check=1  give following results:

[image]

JeanRochCoulon, 2025-01-02:
@AnouarZajni  Improvements have been done. Could you add comments and profiling after fix?   And close it if you think it is worth closing it.

AnouarZajni, 2025-01-08:
Hi,
After analysis, RVFI agent update to monitor/scoreboard only used number of CSRs  will need too much effort with only few secs improvement in runtime. Instead  a new switch on Testbench called tb_performance_mode has been added to disable CSR checks on RVFI which make it faster. This switch is disabling: AXI delays, loggers and rvfi_disable_csr_check ...

yes I'm simulating cv32a65X as per coremark.sh config

AnouarZajni, 2025-01-08:
@JeanRochCoulon please close the issue

closed 2025-01-08 by AnouarZajni

jason23g, 2025-02-07:
Hello @AnouarZajni,

Could you give some guidelines in order to update RVFI agent to monitor/scoreboard only used number of CSRs  and update the spike module in tandem simulation and so on to improve the performance in tandem simulation.

Thanks in advance for your time.

AnouarZajni, 2025-02-11:
Hi @jason23g,
The idea is to upgrade the SPIKE to output a list of used CSRs then use it inside RVFI agent.

Regards,
Anouar

jason23g, 2025-02-11:
Hello @AnouarZajni,

I have tried to upgrade the SPIKE by first changing the CSR_MAX_SIZE to the appropriate then modifying the index of CSR to be from 0 to size and not based on the csr address and to also modify both the SPIKE and spike.sv. However this approach didn't work because as it seems the reference model (aka SPIKE) keeps holding the information of the CSR to the index based on its address. For instance CSR MCAUSE initially is in the index 0x300 and i tried to modify to be in the index 0. Do you have any recommendations to do or how to approach the idea to update RVFI agent to monitor/scoreboard only used number of CSRs.

Thanks in advance for your time.