openhwfoundation/cva6 #2596: [BUG]  Boot linux on Verilator

jason23g, 2024-11-12:
### Is there an existing CVA6 bug for this?

- [X] I have searched the existing bug issues

### Bug Description

Hello guys,

I tried ti run the script linux.sh in the directory verif/regress to boot linux in Verilator, however after 17500s i get a timeout error and there is nothing in my console to indicate that the bootloader (bbl.o)  has even been started. I had searched the existing bugs and the only relative that i could found was that the riscv proxy kernel is not currently supported which is responsible of printing any information in the host terminal. I would like to ask if this is a known bug and if so  could you give some directions to be able to support riscv proxy kernel in the current environment of cva6 and lastly if the riscv proxy kernel would also work in the spike tandem mode.

Thanks in advance for your time.

JeanRochCoulon, 2024-11-15:
Hello @jason23g 
I simulated the first 40M first instructions of Linux with Verilator maybe 2 years ago. To check it, we grepped the presence of some significant program counters in the instruction trace. You may retrieve this work from Github history. 
But we experienced the use of FPGA which is more efficient, and we gave up the simulation way.

JeanRochCoulon, 2024-11-18:
If agreed, this issue can be closed

closed 2024-11-18 by jason23g