openhwfoundation/cva6 #2606: [BUG] Issue with OpenOCD for CVA6-Based Project on Genesys2 FPGA

Yidaen, 2024-11-15:
### Is there an existing CVA6 bug for this?

- [X] I have searched the existing bug issues

### Bug Description

Hello everyone,

I am currently working on a CVA6-based project using the Genesys2 FPGA. I successfully compiled the bitstream and loaded it onto the FPGA. Using UART, I can confirm that the processor is starting up, as indicated by the following output:
```
Hello World!
init SPI
status: 0x0000000000000025
status: 0x0000000000000025
SPI initialized!
initializing SD... 
SD command cmd0         response : 01
SD command cmd55        response : 01
SD command cmd41        response : 01
SD command cmd55        response : 01
SD command cmd41        response : 01
SD command cmd55        response : 01
SD command cmd41        response : 00
sd initialized!
SD card failed!
sd copy return value: FFFFFFFFFFFFFFFE
``` 
Now, I want to run a C program directly on the processor without a Linux operating system. I have previously compiled this program and verified that it works in simulation using Questasim.

To load the program, I installed OpenOCD (https://github.com/riscv-collab/riscv-openocd). Since the Culsans platform does not provide a specific .cfg file, I tried using ariane.cfg, assuming it would work as the Culsans platform uses the Ariane (CVA6) core. However, when running OpenOCD with the following command I encountered the following error:

```
$ openocd -f ariane.cfg

Open On-Chip Debugger 0.12.0+dev-04044-ga2f5da328 (2024-11-15-10:53)
Licensed under GNU GPL v2
For bug reports, read
    http://openocd.org/doc/doxygen/bugs.html
DEPRECATED! use 'adapter speed' not 'adapter_khz'
DEPRECATED! use 'adapter driver' not 'interface'
DEPRECATED! use 'ftdi vid_pid' not 'ftdi_vid_pid'
DEPRECATED! use 'ftdi channel' not 'ftdi_channel'
DEPRECATED! use 'ftdi layout_init' not 'ftdi_layout_init'
DEPRECATED! use 'ftdi layout_signal' not 'ftdi_layout_signal'
Info : auto-selecting first available session transport "jtag". To override use 'transport select <transport>'.
DEPRECATED! use 'gdb report_data_abort', not 'gdb_report_data_abort'
DEPRECATED! use 'gdb report_register_access_error', not 'gdb_report_register_access_error'
Warn : The command 'riscv set_reset_timeout_sec' is deprecated! Please, use 'riscv set_command_timeout_sec'.
ariane.cfg:36: Error: 
in procedure 'script' 
at file "embedded:startup.tcl", line 28
at file "ariane.cfg", line 36
Error: [riscv.cpu] Unsupported DTM version: -1
Error: [riscv.cpu] Could not identify target type.
```

As mentioned, I want to run some compiled C programs on the processor without Linux and, if needed, debug them. Any help or suggestions would be greatly appreciated. Thank you in advance!

Ubuntu 24.04
Genesys2 FPGA

Gchauvon, 2024-11-18:
You must use an older version of OpenOCD or update `ariane.cfg` to not have deprecated command.

Yidaen, 2024-11-18:
I was able to connect by switching to an older version of OpenOCD.

closed 2024-11-18 by Yidaen