openhwfoundation/cva6 #2618: [BUG] Incorrect $fatal call

likeamahoney, 2024-11-20:
### Is there an existing CVA6 bug for this?

- [X] I have searched the existing bug issues

### Bug Description

Hi!
[https://github.com/openhwgroup/cva6/blob/2f81dba77feb67e21340b13dc5ab53f77080970e/core/cache_subsystem/axi_adapter.sv#L204](Here) an example of incorrect call to `$fatal` system task due to SystemVerilog LRM (section 20.10).

As it described in LRM `$fatal` can be called without args or with `finish_number` (`finish_number ::= 0 | 1 | 2`) passed as first arg but not a string message. String message can be passed after the `finish_number`. 

`VCS` don't emit a warning on it but `Siemens Questa` and `Cadence` do. Was this intentional or just a typo?

JeanRochCoulon, 2024-11-20:
Thanks @likeamahoney   I think it is not intentional. I suggest you to submit a PR to fix it

likeamahoney, 2024-11-20:
> Thanks @likeamahoney I think it is not intentional. I suggest you to submit a PR to fix it
Thanks for review!
Done here:
https://github.com/openhwgroup/cva6/pull/2619

commit 25f2f319 (2024-11-20): Fix $fatal system task incorrect usage (#2619)

To fix  #2618
closed 2024-11-20 by JeanRochCoulon