openhwfoundation/cva6 #2620: [BUG] Return address prediction failed

AEzzejjari, 2024-11-20:
### Is there an existing CVA6 bug for this?

- [X] I have searched the existing bug issues

### Bug Description

The PR #2607 contains a scoreboard for the frontend stage. 
Using this PR and the Dhrystone test, we detected that the RAS submodule doesn't pop an address from the stack when a return instruction is triggered. 
In the event of the bug occurring, the RAS stack is full, and the variable in the Frontend module connected to the RAS input pop_data is equal to 1.

Gchauvon, 2025-01-13:
I took your branch and check for this bug. I saw some UVM errors showing a missmatch between the scoreboard implemented in the frontend agent and the RTL. 

```
`UVM_INFO @ 672.000 ns : uvmt_cva6_firmware_test.sv(159) uvm_test_top [TEST] Started RUN
UVM_ERROR @ 42912.000 ns : uvme_cva6_frontend_model.sv(844) uvm_test_top.env.sb.frontend_sb.frontend_model [uvme_cva6_frontend_model_c] ERROR -> the frontend sent wrong kill_s2
UVM_ERROR @ 42912.000 ns : uvme_cva6_frontend_model.sv(846) uvm_test_top.env.sb.frontend_sb.frontend_model [uvme_cva6_frontend_model_c] ERROR -> the frontend sent wrong npc, 0x80003410 != 0x8000553e, npc: 0x80003418, next_predicted_address: 0x0
UVM_ERROR @ 42932.000 ns : uvme_cva6_frontend_model.sv(846) uvm_test_top.env.sb.frontend_sb.frontend_model [uvme_cva6_frontend_model_c] ERROR -> the frontend sent wrong npc, 0x80003418 != 0x80005540, npc: 0x80003420, next_predicted_address: 0x0
UVM_ERROR @ 42952.000 ns : uvme_cva6_frontend_model.sv(846) uvm_test_top.env.sb.frontend_sb.frontend_model [uvme_cva6_frontend_model_c] ERROR -> the frontend sent wrong npc, 0x80003420 != 0x80005548, npc: 0x80003420, next_predicted_address: 0x0
UVM_ERROR @ 42972.000 ns : uvme_cva6_frontend_model.sv(831) uvm_test_top.env.sb.frontend_sb.frontend_model [uvme_cva6_frontend_model_c] ERROR -> Wrong predicted address
UVM_ERROR @ 42972.000 ns : uvme_cva6_frontend_model.sv(846) uvm_test_top.env.sb.frontend_sb.frontend_model [uvme_cva6_frontend_model_c] ERROR -> the frontend sent wrong npc, 0x80003420 != 0x80005548, npc: 0x80003420, next_predicted_address: 0x0
UVM_ERROR @ 42992.000 ns : uvme_cva6_frontend_model.sv(829) uvm_test_top.env.sb.frontend_sb.frontend_model [uvme_cva6_frontend_model_c] ERROR -> Wrong address`
```

However, I didn't see any weird behaviour of the RAS or in the protocol between frontend and Icache. There was no pop or push in the RAS during those UVM errors. 

Do you still have this problem ? If you do, can you give me **logs**, **waveform** and/or **timing** to reproduce and locate this bug.

JeanRochCoulon, 2025-04-10:
The issue was not reproduced on cv32a60x. As this bug impact is on performance and not on functionality, I propose to waive it for cv32a65x by replacing the label cv32a65x by notcv32a65x.