openhwfoundation/cva6 #2626: [BUG] spike : Update CVXIF instruction

AyoubJalali, 2024-11-21:
### Is there an existing CVA6 bug for this?

- [X] I have searched the existing bug issues

### Bug Description

Hello, as we know the CVXIF protocol has been changed so the verification should be updated based on the new spec, also should update spike to support [new custom instruction](https://github.com/openhwgroup/cva6/blob/master/verif/env/uvme/cvxif_vseq/custom_instructions_cvxif_1_0_0.rst), I notice that @Gchauvon has update some instruction in spike but not all of them.

zchamski, 2025-01-08:
This is work in progress (local commits done).  Due to the scope of changes it is a major update: there are new instruction formats that need to be decoded/handled.

zchamski, 2025-01-23:
Implementation available in https://github.com/openhwgroup/core-v-verif/pull/2575, under test by @AyoubJalali.

zchamski, 2025-03-18:
@AyoubJalali, at the current stage of development I propose to close this issue and open specific ones on implementation bugs (incorrect result values in instructions etc.)

AyoubJalali, 2025-03-21:
was resolved and open an other issue #2852

closed 2025-03-21 by AyoubJalali