openhwfoundation/cva6 #2645: [BUG] Result of riscv_unaligned_load_store_test smoke-gen test is inconsistent

valentinThomazic, 2024-12-03:
### Is there an existing CVA6 bug for this?

- [X] I have searched the existing bug issues

### Bug Description

The result of riscv_unaligned_load_store_test is inconsistent.
The test completes in all cases but the result will sometimes be a failure.

The issue seems to be on master branch (currently b5b316ad1209ab7f8977429063481e181f007f65) and can be seen on the dashboard report of this PR : https://github.com/openhwgroup/cva6/pull/2641 (see pipeline 48858 on the [CVA6 dashboard](https://riscv-ci.pages.thales-invia.fr/dashboard/)) which has no reason to impact the results of tests executions.

I also saw that on latest (https://github.com/openhwgroup/cva6/pull/2528/commits/ff70060a23789ed0b2346e5fc0c5c2e33f988b91) pipeline of https://github.com/openhwgroup/cva6/pull/2528 (pipeline 48965 on the [CVA6 dashboard](https://riscv-ci.pages.thales-invia.fr/dashboard/)) though it is harder to say for sure since this PR changes RTL. Running the test another time gives a successful result.

valentinThomazic, 2024-12-03:
It seems the issue happens between the end of test and the end of simulation.
it seems that the tests finishes successfully and the correct instructions are executed but this error sometimes occurs:
<pre>UVM_INFO @ 672.000 ns : uvmt_cva6_firmware_test.sv(137) uvm_test_top [TEST] Started RUN
"/gitlab-runner/runner_riscv-public/builds/yD5zmwgi3/7/riscv-ci/cva6/core/scoreboard.sv", 346: uvmt_cva6_tb.cva6_dut_wrap.cva6_tb_wrapper_i.i_cva6.issue_stage_i.i_scoreboard.genblk11[1].genblk1[2].unnamed$$_0: started at 48512000ps failed at 48512000ps
	Offending '(trans_id_i[1] != trans_id_i[2])'
Fatal: "/gitlab-runner/runner_riscv-public/builds/yD5zmwgi3/7/riscv-ci/cva6/core/scoreboard.sv", 346: uvmt_cva6_tb.cva6_dut_wrap.cva6_tb_wrapper_i.i_cva6.issue_stage_i.i_scoreboard.genblk11[1].genblk1[2].unnamed$$_0: at time 48512000 ps
Two or more functional units are retiring instructions with the same transaction id!
$finish called from file "/gitlab-runner/runner_riscv-public/builds/yD5zmwgi3/7/riscv-ci/cva6/core/scoreboard.sv", line 346.

uvmt_cva6_tb.end_of_test: *** Test Summary ***

    FFFFFFFF   AAAAAA   IIIIII  LL        EEEEEEEE  DDDDDDD       
    FF        AA    AA    II    LL        EE        DD    DD      
    FF        AA    AA    II    LL        EE        DD    DD      
    FFFFF     AAAAAAAA    II    LL        EEEEE     DD    DD      
    FF        AA    AA    II    LL        EE        DD    DD      
    FF        AA    AA    II    LL        EE        DD    DD      
    FF        AA    AA  IIIIII  LLLLLLLL  EEEEEEEE  DDDDDDD       
    ----------------------------------------------------------
                   SIMULATION FAILED - ABORTED              
    ----------------------------------------------------------
$finish at simulation time 48512.000 ns</pre>

JeanRochCoulon, 2025-01-02:
@AyoubJalali can you have a look at this ?

AyoubJalali, 2025-01-02:
> @AyoubJalali can you have a look at this ?

This is a RTL bug, an assertion failed for some reason that i don't know to be honest
I have opened a [similar issue](https://github.com/openhwgroup/cva6/issues/2376)

ASintzoff, 2025-01-08:
Close as duplicate of #2376

closed 2025-01-08 by ASintzoff