openhwfoundation/cva6 #2656: [BUG] Incorrect handling of PMPADDR register reads in NA4 / NAPOT address matching modes

zchamski, 2024-12-09:
### Is there an existing CVA6 bug for this?

- [X] I have searched the existing bug issues

### Bug Description

The RTL code of CSR read process (https://github.com/openhwgroup/cva6/blob/master/core/csr_regfile.sv#L851C1-L857C71) incorrectly assumes that the last bit of `pmpaddr` registers should be read as 1 if the corresponding `pmpNcfg.A[1]` bit is set.  Table 19 in the privileged ISA spec (v20240411) states that the value of bit 0 in a `pmpaddrN` register **can** be 0 in `NA4` mode and **must** be 0 in `NAPOT` mode with 8 byte ranges:

[image]

@Moschn, can you please have a look at this?

Moschn, 2024-12-09:
I think you are correct. I'm not sure where this bug crept in; at some point, this was working correctly (I checked the supported granularity in software a couple years ago). Feel free to propose a fix!

Moschn, 2025-01-17:
I checked this again today. In the past, we did not support NA4, which coincides with the current behavior of the code. To clarify, we set G=2 (Granularity of a minimum of $2^{G+2} = 16$ bytes) and thus NA4 to be disabled. The important sentences in the spec:

>  When $G>=1$, the NA4 mode is not selectable. When $G>=2$ and $pmpcfg_i.A[1]$ is set, i.e. the mode is NAPOT, then bits $pmpaddr_i[G-2:0]$ read as all ones. When $G>=1$ and $pmpcfg_i.A[1]$ is clear, i.e. the mode is OFF or TOR, then bits $pmpaddr_i[G-1:0]$ read as all zeros.

The current code performs exactly this behavior:
```verilog
if (pmpcfg_q[index].addr_mode[1] == 1'b1) // i.e., A[1] is set
  csr_rdata = {pmpaddr_q[index][CVA6Cfg.PLEN-3:1], 1'b1}; // pmpaddr_i[0:0] = 1
else csr_rdata = {pmpaddr_q[index][CVA6Cfg.PLEN-3:1], 1'b0}; // A[1] is not set -> pmpaddr_i[1:0] = 0
```

I was also confused at first when I reread the spec and the code, but I think there was a mistake that I thought G=2 means 8-byte granularity. Instead, G=2 means 16-byte granularity, and then also the table is correct, i.e., in NAPOT mode, always force bit 0 to 1.

I believe this still holds with respect to the recent changes (#2685).

kaci1928, 2026-10-28:
I can still reproduce this on current master.

Affected commit: `742dc82a0c706e0a3bfb98b709fd3ee06d50d12a`
Target: `cv64a6_imafdc_sv39_hpdcache`

This configuration uses PMP granularity G=1 and enables NAPOT.

Minimal reproduction:

```asm
csrw pmpcfg0, zero
csrw pmpaddr0, zero
li   t0, 0x18              # NAPOT
csrw pmpcfg0, t0
csrr t2, pmpaddr0
```

Observed behavior:

```text
CVA6:  pmpaddr0 = 1
Spike: pmpaddr0 = 0
```

For G=1 and NAPOT, the privileged ISA only requires the low bits to read as all ones when G>=2. Therefore bit 0 should retain the stored value in this case and should read as zero here.

The current implementation forces bit 0 to one whenever NA4/NAPOT is selected:

- `core/csr_regfile.sv`, PMPADDR read path around lines 995-1001
- `core/cva6_rvfi.sv`, RVFI mirror around lines 661-662

The same issue was previously discussed in #697 and fixed by #698, but the current code appears to have regressed to forcing `1'b1`.

This reproduction only demonstrates an incorrect CSR readback and RVFI mirror. The underlying PMP matching logic uses the raw stored address, so I have not observed evidence of a PMP access-control bypass.