openhwfoundation/cva6 #2657: [BUG] [CV32A65X] Writes into PMPADDR registers do not follow specification

zchamski, 2024-12-09:
### Is there an existing CVA6 bug for this?

- [X] I have searched the existing bug issues

### Bug Description

Since the CV32A65X specification has been updated to make bit 0 of `pmpaddrN` registers read-only zero (see https://github.com/openhwgroup/cva6/pull/2651), the implementation of `pmpaddrN` CSR writes is not in line with the spec.  The incriminated code is https://github.com/openhwgroup/cva6/blob/master/core/csr_regfile.sv#L1710C1-L1713C71.

JeanRochCoulon, 2024-12-09:
Fixed by #2658

JeanRochCoulon, 2025-01-02:
@zchamski We are cleaning the "Github Issues". I assume this issue is closed. May I ask you to close it if it is the case?

JeanRochCoulon, 2025-01-07:
This GitHub issue has not been fixed by #2685.