openhwfoundation/cva6 #2668: CVA6 Genus Support

rsar97, 2024-12-15:
Hi,

I am trying to synthesize CVA6 using the Genus. There are no compilation errors with Xcelium. However, there are multiple errors when I pass it through Cadence Genus?  Is CVA6 supported on Cadence Genus ? 

    string f_name;
         |
Error   : Unsupported declaration. [VLOGPT-32] [read_hdl]
        : String type in file '/home/rsaravan/postfuzz/cva6/viv.sv' on line 8527, column 10.
        : The following declarations are not supported for synthesis:
    - tri0
    - tri1
    - event
    - class
    - chandle
    - SV string type
    - associative array.
    string f_name;
         |
Error   : Illegal use of SystemVerilog reserved word. [VLOGPT-8] [read_hdl]
        : Keyword 'string' in file '/home/rsaravan/postfuzz/cva6/viv.sv' on line 8527, column 10.
        : SystemVerilog introduces many new keywords.  They are:
    alias           do              join_any        solve
    always_comb     endclass        join_none       static
    always_ff       endclocking     local           string
    always_latch    endinterface    logic           struct
    assert          endprogram      longint         super
    assert_strobe   endproperty     modport         this
    before          endsequence     new             throughout
    bind            enum            null            timeprecision
    bit             export          packed          timeunit
    break           extends         priority        type
    byte            extern          program         typedef
    chandle         final           property        union
    class           first_match     protected       unique
    clocking        forkjoin        pure            unique0
    const           iff             rand            var
    constraint      import          randc           virtual
    context         inside          ref             void
    continue        int             sequence        wait_order
    cover           interface       shortint        with
    dist            intersect       shortreal       within
To use these keywords as identifiers in a design, you must read the input design with 'read_hdl -v1995' or 'read_hdl -v2001'; or use the pragma:
    `begin_keywords "1364-1995"
or    `begin_keywords "1364-2001".
    string f_name;
                |
Error   : Parsing error. [VLOGPT-1] [read_hdl]
        : Expecting token ';', found '<identifier>' in file '/home/rsaravan/postfuzz/cva6/viv.sv' on line 8527, column 17.
        : Invalid Verilog syntax is parsed, or unsupported Verilog syntax is encountered.
    if ($value$plusargs("checkpoint=%s", f_name)) begin
                                         |
Error   : Reference to undeclared variable. A variable/parameter must be declared before it is referenced. [VLOGPT-20] [read_hdl]
        : Symbol 'f_name' in file '/home/rsaravan/postfuzz/cva6/viv.sv' on line 8534, column 42.
        : Verilog module is using a parameter that is not defined in the module. You will get a notification for this, if you read the verilog file using the read_hdl command.
    if ($value$plusargs("checkpoint=%s", f_name)) begin
                                               |
Warning : Unsupported system task or function: assuming value 1'b1. [VLOGPT-31]
        : System function '$value$plusargs' in file '/home/rsaravan/postfuzz/cva6/viv.sv' on line 8534, column 48.
        : The $signed and $unsigned system functions are always supported, and synthesizable SystemVerilog system functions are supported when SystemVerilog parsing is enabled.  No other system functions are supported.
      hex_file = $fopen({f_name,".mainram.hex"}, "r");
                         |
Error   : Reference to undeclared variable. A variable/parameter must be declared before it is referenced. [VLOGPT-20] [read_hdl]
        : Symbol 'f_name' in file '/home/rsaravan/postfuzz/cva6/viv.sv' on line 8535, column 26.
      hex_file = $fopen({f_name,".mainram.hex"}, "r");
                                                    |
Warning : Unsupported system task or function: assuming value 1'b1. [VLOGPT-31]
        : System function '$fopen' in file '/home/rsaravan/postfuzz/cva6/viv.sv' on line 8535, column 53.
      while (!$feof(hex_file)) begin
                            |
Warning : Unsupported system task or function: assuming value 1'b1. [VLOGPT-31]
[... 57 lines cut ...]
endmodule
        |
Error   : Parsing error. [VLOGPT-1] [read_hdl]
        : in file '/home/rsaravan/postfuzz/cva6/viv.sv' on line 15338, column 9.
endmodule
        |
Error   : Parsing error. [VLOGPT-1] [read_hdl]
        : Expecting token ';', found 'endmodule' in file '/home/rsaravan/postfuzz/cva6/viv.sv' on line 15338, column 9.
  localparam int unsigned EXP_BITS = fpnew_pkg::exp_bits(FpFormat);
                                                         |
Error   : Reference to undeclared variable. A variable/parameter must be declared before it is referenced. [VLOGPT-20] [read_hdl]
        : Symbol 'FpFormat' in file '/home/rsaravan/postfuzz/cva6/viv.sv' on line 15363, column 58.
  localparam int unsigned EXP_BITS = fpnew_pkg::exp_bits(FpFormat);
                                                         |
Error   : Reference to undeclared variable. A variable/parameter must be declared before it is referenced. [VLOGPT-20] [read_hdl]
        : Symbol 'FpFormat' in file '/home/rsaravan/postfuzz/cva6/viv.sv' on line 15363, column 58.
  localparam int unsigned EXP_BITS = fpnew_pkg::exp_bits(FpFormat);
                                                         |
Error   : Reference to undeclared variable. A variable/parameter must be declared before it is referenced. [VLOGPT-20] [read_hdl]
        : Symbol 'FpFormat' in file '/home/rsaravan/postfuzz/cva6/viv.sv' on line 15363, column 58.
  localparam int unsigned MAN_BITS = fpnew_pkg::man_bits(FpFormat);
                                 |
Error   : Parsing error. [VLOGPT-1] [read_hdl]
        : Expecting token ';', found '<identifier>' in file '/home/rsaravan/postfuzz/cva6/viv.sv' on line 15364, column 34.
    logic [EXP_BITS-1:0] exponent;

JeanRochCoulon, 2024-12-17:
Hello,  VCS from Synopsys and Verilator are suported. So far, some CVA6 users use Cadence on their own, but it is not fully functional with repository scripts/Makefile. But the door is open to contribute !

JeanRochCoulon, 2025-01-02:
@rsar97 We are cleaning the "Github Issues". I assume this issue is closed. May I ask you to close it ?

closed 2025-01-02 by rsar97