openhwfoundation/cva6 #2702: [Simulation] Segmentation fault while running smoke-gen_tests.sh

khandelwaltanuj, 2025-01-13:
### Is there an existing CVA6 bug for this?

- [X] I have searched the existing bug issues

### Bug Description

Hello 

I am getting a segmentation fault while running the script source verif/regress/smoke-gen_tests.sh 
VCS version used is this one: U-2023.03-SP1 (I am wondering if the version of vcs used is not correct) 

Following is the list of different configuration used: 
Mon, 13 Jan 2025 10:25:22 INFO     GCC Version: 15.0.0
Mon, 13 Jan 2025 10:25:22 INFO     Spike Version: 1.1.1-dev 464bf7a0
Mon, 13 Jan 2025 10:25:23 INFO     Verilator Version: Verilator 5.008 2023-03-04 rev v5.008 (mod)
Mon, 13 Jan 2025 10:25:23 INFO     Creating output directory: /home/560.1.429-MOSAIC/users/tk223379/cva6/verif/sim/out_2025-01-13

Mon, 13 Jan 2025 10:25:23 INFO     Iteration number: 1
Mon, 13 Jan 2025 10:25:23 INFO     Processing regression test list : cva6_base_testlist.yaml, test: riscv_unaligned_load_store_test
Mon, 13 Jan 2025 10:25:23 INFO     Found matched tests: riscv_unaligned_load_store_test, iterations:1
Mon, 13 Jan 2025 10:25:23 INFO     CVA6 Configuration is  and target is cv32a65x

Following is the snippet of segmentation fault. 
UVM_INFO @ 32.000 ns : uvma_cvxif_drv.sv(218) uvm_test_top.env.cvxif_agent.driver [CVXIF_DRV] CVXIF Agent is disabled : All CPU Requests are rejected !!
UVM_INFO @ 672.000 ns : uvmt_cva6_firmware_test.sv(137) uvm_test_top [TEST] Started RUN

An unexpected termination has occurred in /home/PROJECT/cva6/verif/sim/vcs_results/default/vcs.d/simv due to a signal: Segmentation fault

During Functional coverage runtime at /home/PROJECT/cva6/verif/sim//../core-v-verif/lib/uvm_agents/uvma_isacov/uvma_isacov_pkg.sv, 22
 in covergroup uvma_isacov_pkg::cg_csrtype declared at /home/PROJECT/cva6/verif/sim//../core-v-verif/lib/uvm_agents/uvma_isacov/cov/uvma_isacov_cov_model.sv, 1008
 
Command line: /home/PROJECT/cva6/verif/sim/vcs_results/default/vcs.d/simv ++/home/PROJECT/cva6/verif/sim/out_2025-01-13/asm_tests/riscv_unaligned_load_store_test_0.o +elf_file=/home/PROJECT/cva6/verif/sim/out_2025-01-13/asm_tests/riscv_unaligned_load_store_test_0.o +core_name=cv32a65x +config_file=/home/PROJECT/cva6/config/gen_from_riscv_config/cv32a65x/spike/spike.yaml +tohost_addr=80009cc0 +signature=/home/PROJECT/cva6/verif/sim/out_2025-01-13/asm_tests/riscv_unaligned_load_store_test_0.o.signature_output +UVM_TESTNAME=uvmt_cva6_firmware_test_c +report_file=/home/PROJECT/cva6/verif/sim/out_2025-01-13/vcs-uvm_sim/riscv_unaligned_load_store_test_0.cv32a65x.log.yaml +core_name=cv32a65x +UVM_VERBOSITY=UVM_NONE +debug_disable=1+UVM_VERBOSITY=UVM_NONE +ntb_random_seed=1 -sv_lib /home/PROJECT/cva6/tools/spike/lib/libcustomext -sv_lib /home/PROJECT/cva6/tools/spike/lib/libyaml-cpp -sv_lib /home/PROJECT/cva6/tools/spike/lib/libriscv -sv_lib /home/PROJECT/cva6/tools/spike/lib/libfesvr -sv_lib /home/PROJECT/cva6/tools/spike/lib/libdisasm +vcs+lic+wait +debug_disable=1+UVM_VERBOSITY=UVM_NONE +ntb_random_seed=1 +tandem_enabled=0 +debug_disable=1+UVM_VERBOSITY=UVM_NONE +ntb_random_seed=1

--- Stack trace follows:

Dumping VCS Annotated Stack:
#0  0x0000155547f1de0b in waitpid () from /usr/lib64/libc.so.6
#1  0x0000155547e71747 in do_system () from /usr/lib64/libc.so.6
#2  0x000015555221e5c0 in SNPSle_10ee25eff68cd8461c9146fa1d0b35e87067f3c8015b313e639d2928478c79b3f673f99203bcf8be64600612100082236bffb2007f1e0ef9 () from /home/cao/synopsys/VCS/U-2023.03-SP1/linux64/lib/liberrorinf.so
#3  0x000015555222011c in SNPSle_10ee25eff68cd8461c9146fa1d0b35e87067f3c8015b313efba706aab251478fa49e66610e453774633a6c152e7ef778f93045171fb3645d () from /home/cao/synopsys/VCS/U-2023.03-SP1/linux64/lib/liberrorinf.so
#4  0x0000155552218363 in SNPSle_d35ca1ff70d465c2b9b1a72eee90a50630165806651fae96c1bbda5b5d02066c () from /home/cao/synopsys/VCS/U-2023.03-SP1/linux64/lib/liberrorinf.so
#5  0x000015554d76911b in SNPSle_64133461705005bb725549e2e6fa1b3f () from /home/cao/synopsys/VCS/U-2023.03-SP1/linux64/lib/libvcsnew.so
#6  0x000015554d3e1bad in SNPSle_82244d58c54c18c70d63edc9becab634 () from /home/cao/synopsys/VCS/U-2023.03-SP1/linux64/lib/libvcsnew.so
#8  0x000015554f1f1fb1 in SNPSle_c556e4f00792e3f14dc9ecae412cc5b0ab24f20aa63fbd24 () from /home/cao/synopsys/VCS/U-2023.03-SP1/linux64/lib/libvcsnew.so
#9  0x000015554f158444 in SNPSle_942396d4317cf70c2ac181d9a2c10e589d918bb0a6c7d9711132598284364c5b7c7a5cf4452ff3a5a69c478261981b60d953717387fa4117 () from /home/cao/synopsys/VCS/U-2023.03-SP1/linux64/lib/libvcsnew.so
#10 0x000015554f17aa9c in SNPSle_737d9de40279cd1ec3f943b10477b5cca65bdd9f8ad0b726ad7eaa4e9e74652c734d64a1616f5ec2 () from /home/cao/synopsys/VCS/U-2023.03-SP1/linux64/lib/libvcsnew.so
#11 0x000015554f17b9e3 in SNPSle_507928f2ab5d74e3f9fa4c2838b9637c () from /home/cao/synopsys/VCS/U-2023.03-SP1/linux64/lib/libvcsnew.so
#12 0x000015554f181331 in SNPSle_e8f6b361da6190009f37a52391ad053dd4df9fd6c14f8bbe3cde92ff598014c7 () from /home/cao/synopsys/VCS/U-2023.03-SP1/linux64/lib/libvcsnew.so
#13 0x000015554f19dc42 in SNPSle_50c646f9dc9b52d8cd06573e24e50c6f43b5b7b03b938b16 () from /home/cao/synopsys/VCS/U-2023.03-SP1/linux64/lib/libvcsnew.so
#14 0x000015554d30bef6 in SNPSle_3d00fd0ec33c85785d6dc9c72a6c0c30 () from /home/cao/synopsys/VCS/U-2023.03-SP1/linux64/lib/libvcsnew.so
#15 0x00001555532fd469 in T_VCSgd_Zrgws_24_0 () from /home/PROJECT/cva6/verif/sim/vcs_results/default/vcs.d/simv.daidir/_prev_archive_1.so
#16 0x00001555532c3ec7 in F_VCSgd_Zrgws_213_0 () from /home/PROJECT/cva6/verif/sim/vcs_results/default/vcs.d/simv.daidir/_prev_archive_1.so
#17 0x00001555532c11b1 in F_VCSgd_Zrgws_210_0 () from /home/PROJECT/cva6/verif/sim/vcs_results/default/vcs.d/simv.daidir/_prev_archive_1.so
#18 0x000015554d806991 in SNPSle_ba9d6ef1a21efcfe () from /home/cao/synopsys/VCS/U-2023.03-SP1/linux64/lib/libvcsnew.so
#19 0x000015554d5dfeb2 in SNPSle_63c7abb936d47b07d29d61775f9dbe8c1486beb9213552a3 () from /home/cao/synopsys/VCS/U-2023.03-SP1/linux64/lib/libvcsnew.so
#20 0x000015554d780fdf in SNPSle_490598bfebcc8e81efcf3d1815066cc9 () from /home/cao/synopsys/VCS/U-2023.03-SP1/linux64/lib/libvcsnew.so
#21 0x000015554d79dc1e in SNPSle_490598bfebcc8e81 () from /home/cao/synopsys/VCS/U-2023.03-SP1/linux64/lib/libvcsnew.so
#22 0x000000000040eb43 in main ()

With Best Regards
Tanuj Khandelwal

JeanRochCoulon, 2025-01-13:
Hi @khandelwaltanuj 
Yes, the VCS version is critical to make UVM work.
We are using V-2023.12-SP1
Can you try with this version?
Regards

khandelwaltanuj, 2025-01-13:
Hi @JeanRochCoulon 

I believe we do not have this version. I will se if we can get this version and I will come back to you.

Regards
Tanuj

khandelwaltanuj, 2025-01-21:
Hello @JeanRochCoulon ,

With this version the gen smoke seems to be passing. 
I seems to be getting warning during make clean. 

Tue, 21 Jan 2025 16:33:12 INFO     Target: cv32a65x
Tue, 21 Jan 2025 16:33:12 INFO     ISA rv32imc_zba_zbb_zbc_zbs_zcb_zicsr_zifencei
Tue, 21 Jan 2025 16:33:12 INFO     vcs-uvm sim log dir: /home/560.1.429-MOSAIC/users/tk223379/cva6/verif/sim/out_2025-01-21/vcs-uvm_sim
Tue, 21 Jan 2025 16:33:13 INFO     Running vcs-uvm sim: /home/560.1.429-MOSAIC/users/tk223379/cva6/verif/sim/out_2025-01-21/asm_tests/riscv_unaligned_load_store_test_0.o
Makefile:14: must set CVA6_REPO_DIR to point at the root of CVA6 sources and CVA6_TB_DIR to point here -- doing it for you...
basename: missing operand
Try 'basename --help' for more information.
[VCS] Cleanup (entire vcs_work dir)
rm -rf /home/560.1.429-MOSAIC/users/tk223379/cva6/verif/sim/vcs_results/ verdiLog/ simv* *.daidir *.vpd *.fsdb *.db csrc ucli.key vc_hdrs.h novas* inter.fsdb uart
rm -f *.txt
rm -f trace*.log
rm -f trace*.dasm
rm -f *.vpd *.fsdb *.vcd *.fst

Regards
Tanuj

closed 2025-01-22 by khandelwaltanuj