openhwfoundation/cva6 #2715: Readme instructions - smoke-tests.sh unavailable

pcotret, 2025-01-17:
In the readme, instructions sometimes talk about :
```bash
bash verif/regress/smoke-tests.sh
```
`smoke-tests.sh` doesn't exist. However, we have a few files depending on the CPU variant we want to use: https://github.com/openhwgroup/cva6/tree/master/verif/regress

isaar-ahmad, 2025-01-20:
It seems that the original smoke-tests.sh was split into variant-specific smoke-test scripts in October 2023.

See commit 8a457272b72641ef2fdc87d4c8a4d2765be1f0f4 for details.

You may want to try the following and see if it works : 
`
bash verif/regress/smoke-tests-cv32a65x.sh # 32-bit variant
bash verif/regress/smoke-tests-cv64a6_imafdc_sv39.sh # 64-bit variant
`
I have been able to work with the 64-bit variant in this manner.

JeanRochCoulon, 2025-01-22:
A fix would be welcomed !

pcotret, 2025-01-22:
@JeanRochCoulon That's what I was thinking about ;)
I'll do it asap

closed 2025-01-23 by pcotret