openhwfoundation/cva6 #2734: [BUG] Optional Zifencei extension

ASintzoff, 2025-01-23:
### Is there an existing CVA6 bug for this?

- [x] I have searched the existing bug issues

### Bug Description

The extension Zifencei (`fence.i` instruction) shall be optional.

Therefore, a parameter `RVZifencei` shall be defined to configure it at RTL level. The extension shall be disabled on CV32A60X and CV32A65X (@yanicasa). 

A way to disable the extension (`fence.i` instruction) shall be provided by Spike. (@zchamski)

zchamski, 2025-01-23:
Spike implementation of `fence.i` assumes that `Zifencei` standard *extension* is always present: it does not have the appropriate `require(EXT_....)` clause in the instruction implementation of `fence.i`.

Spike TODOs:

* Ensure that the proper flag for **optional** presence of `Zifencei` exists (implement one if not.)
* Add appropriate guard to `fence.i` implementation.
* Update ISA strings where appropriate.
* Test.

zchamski, 2025-03-18:
The Zifencei extension is now disabled for CV32A6*X and the optional feature is merged and deployed in CI.

@ASintzoff : I think you can close this issue.   If Zifencei-related problems appear on non-CV32A6*X cores, they should be filed as separate, non-CV32A65X issues.

closed 2025-03-18 by ASintzoff