openhwfoundation/cva6 #2737: Add readthedocs action to GitHub

cathales, 2025-01-23:
As discussed in https://github.com/openhwgroup/cva6/issues/2650, a GitHub Action could be added to check that the docs can be built before merging pull requests.

There is an official repository: https://github.com/readthedocs/actions

I’m not sure if it is what we want to do.

valentinThomazic, 2025-02-17:
Here is what I proposed on [the previous issue](https://github.com/openhwgroup/cva6/issues/2650)

> There is an option in Read the Docs enabling the build on each pull request as a Github action. This would not only make debugging easier but would also prevent merging commits that break the documentation build while providing a preview of what is to be merged when the build is successful.
> 
> Here is the (read the docs) documentation about this feature: [feature](https://docs.readthedocs.io/en/stable/pull-requests.html) [feature configuration](https://docs.readthedocs.io/en/stable/guides/pull-requests.html)
> 
> Of course, it is up to OpenHW Group to decide wether they want to do that or not, since this would trigger more builds.

cathales, 2025-02-17:
If I understand it correctly, Valentin’s suggestion does not require adding an action yaml file in the repository.
It seems to be a more robust solution as the `actions` repositories are not always well maintained (we already had issues with verible-format)…

However, it can only be enabled on OpenHW Group side, with an access to their ReadTheDocs account.

Would you have such an access @MikeOpenHWGroup?
Here is the link that Valentin had provided to enable this feature: https://docs.readthedocs.io/en/stable/guides/pull-requests.html