openhwfoundation/cva6 #2741: g++ error while running the smoke tests

RakeshpatilLB, 2025-01-27:
[image]

JeanRochCoulon, 2025-01-28:
This is a Verilator build error, not a CVA6 issue. Please check you compile the Verilator version recommended by README, then check your terminal env.

JeanRochCoulon, 2025-01-28:
If this issue is fixed, could you close the issue (with the rational) ?

RakeshpatilLB, 2025-01-28:
Actually, I am unable to determine where it is going wrong. I have tried to clone and build the toolchain from scratch multiple times, but I keep encountering this error. I have set up the repository on WSL2 (Ubuntu). Is there a chance that WSL2 could be the reason for the error?

closed 2025-03-05 by github-actions[bot]

RakeshpatilLB, 2025-03-10:
There can be multiple reasons behind this issue. The issue i encountered is due to insufficient ram. The processes related to building of verilator get killed as soon as the ram gets filled completely.