openhwfoundation/cva6 #2746: Steps to run custom assembly programs

RakeshpatilLB, 2025-01-28:
I have successfully installed all the CVA6 variants, and the smoke tests completed successfully. However, I am unsure about the steps required to run custom assembly programs on a specific CVA6 variant. Could you please provide guidance or a detailed process to achieve this?

RakeshpatilLB, 2025-02-16:
Got a solution for this issue, not sure if it is best/correct ,but it worked.

step1: arrange your custom assembly code in the given test template ( custom_test_template.S).
step2:  replace --c_tests with --asm_tests, replace the .c file name with the .S file 

source verif/sim/setup-env.sh
export DV_SIMULATORS=veri-testharness
cd ./verif/sim

python3 cva6.py --target cv32a60x --iss=$DV_SIMULATORS --iss_yaml=cva6.yaml \
--asm_tests ../tests/custom/hello_world/custom_test_template.S \
--linker=../../config/gen_from_riscv_config/linker/link.ld \
--gcc_opts="-static -mcmodel=medany -fvisibility=hidden -nostdlib \
-nostartfiles -g ../tests/custom/common/syscalls.c \
../tests/custom/common/crt.S -lgcc \
-I../tests/custom/env -I../tests/custom/common"

Pls let me know if anyone finds out a better way than this.

valentinThomazic, 2025-02-24:
This is the correct way of doing so. If you want to run it automatically you could create a shell script to automate it.
If you want to run several test you can also create a test list.

closed 2025-04-01 by github-actions[bot]