openhwfoundation/cva6 #2754: cv64a6_imafdc_sv39: Issues while running 64 bit configuration of smoke_

khandelwaltanuj, 2025-01-31:
### Is there an existing CVA6 bug for this?

- [x] I have searched the existing bug issues

### Bug Description

I am trying to run the CVA6 smoke-gen test with  64 bit configuration (cv64a6_imafdc_sv39) and I am seeing many warnings of connection size mismatch in VCS and an AXI assertion fail (RLEN should be < 1). I am wondering if I am missing some configuration somewhere. 
I am running on master branch: 

Here are some of the  warnings and error:

"tags_q[0].is_page[((32'b00000000000000000000000000000011 - 1) - 0)][0]"
  The select index is out of declared bounds : [1:0].
  In module instance : i_itlb i_dtlb 
  In module : cva6_tlb.

"issue_instr_i[1]"
  The select index is out of declared bounds : [0:0].
  In module instance : i_issue_read_operands 
  In module : issue_read_operands.

AXI assertion error. 
148: uvmt_cva6_tb.cva6_dut_wrap.axi_assert.cva6_axi_assert.cva6_arlen: started at 1352000ps failed at 1352000ps
	Offending '((axi_assert_if.ar_len == 0) || (axi_assert_if.ar_len == 1))'
UVM_ERROR @ 1352.000 ns : uvmt_cva6_axi_assert.sv(149) reporter [ AXI4 protocol checks assertion ] Violation of AXI4_CVA6_ARLEN

I can see in the uvmt_cva6_tb.sv following hard coded variables. I am wondering if they need to be changed when we shift to 64 bit and may be some other settings need to be done that I may be missing ? 
   // ENV (testbench) parameters
   parameter int ENV_PARAM_INSTR_ADDR_WIDTH  = 32;
   parameter int ENV_PARAM_INSTR_DATA_WIDTH  = 32;
   parameter int ENV_PARAM_RAM_ADDR_WIDTH    = 22;

Thanks and Regards
Tanuj Khandelwal

JeanRochCoulon, 2025-01-31:
Be aware that vcs-uvm target is dedicated to cv32a60x or cv32a65x, not 64 bits configuration.
To adapt it for 64bits, of course Spike and UVM configuration need to be adapted.

khandelwaltanuj, 2025-01-31:
Thanks 

I have just this variable: 
bash-4.4$ echo $DV_TARGET 
cv64a6_imafdc_sv39

Which, I assume is not enough. I am wondering if I can find some information somewhere ? 

Regards
Tanuj

khandelwaltanuj, 2025-01-31:
By the way, I ran this script
source verif/regress/smoke-tests-cv64a6_imafdc_sv39.sh 

The tests seems to be passing but warnings related to signals are still there.

closed 2025-03-03 by khandelwaltanuj

Ly-Van-Minh, 2025-12-01:
Hi @JeanRochCoulon 

Same issue for cva6_shared_tlb.sv 

[image]

[image]